superna9999/meta-meson #2: Trouble booting yocto image from uboot

bjharper, 2017-04-18:
Hi Neil,
   I've completed a custom yocto build using your meta-meson but im having some issues getting it to boot via u-boot.

I see the following output in uboot and then it appears to hang...

`odroidc2#setenv bootcmd setenv bootargs console=ttyAML0,115200 consoleblank=0 root=/dev/mmcblk0p2 rootwait rw\;mmc info\;fatload mmc 1 0x1000000 meson-gxbb-odroidc2.dtb\;fatload mmc 1  0x01080000 uImage\;bootm 0x01080000 - 0x1000000

odroidc2#run bootcmd
Device: SDIO Port C
Manufacturer ID: 15
OEM: 100
Name: AGND3 
Tran Speed: 52000000
Rd Block Len: 512
MMC version 5.0
High Capacity: Yes
Capacity: 14.6 GiB
Bus Width: 8-bit
ret = 1 .[mmc_init] mmc init success
reading meson-gxbb-odroidc2.dtb
14949 bytes read in 6 ms (2.4 MiB/s)
reading uImage
6856776 bytes read in 333 ms (19.6 MiB/s)
ee_gate_off ...
## Booting kernel from Legacy Image at 01080000 ...
   Image Name:   Poky (Yocto Project Reference Di
   Image Type:   AArch64 Linux Kernel Image (uncompressed)
   Data Size:    6856712 Bytes = 6.5 MiB
   Load Address: 01080000
   Entry Point:  01080000
   Verifying Checksum ... OK
load dtb from 0x1000000 ......
## Flattened Device Tree blob at 01000000
   Booting using the fdt blob at 0x1000000
   Loading Kernel Image ... OK
   kernel loaded at 0x01080000, end = 0x0170a008
   Loading Device Tree to 000000001fff9000, end 000000001ffffa64 ... OK

Starting kernel ...

uboot time: 28678927 us
`
I dont see anything further on console (or hdmi) but heartbeat light is flashing.

I have triied both amlogic-s905 and hardkernel-odroidc2 version of the build and tried with hardkernel u-boot and also an upstream version, both with the same issue.

If you have the time, do you have any thoughts on what might be the problem?, maybe I have a build issue somewhere.

Many Thanks,

Brad.

superna9999, 2017-04-18:
Hi Brad,

On which revision did you test ? I just pushed the 4.10 stable version but did not had time to really test it on those devices.

If heartbeat is flashing, it means the kernel has started, but I have issued with Yocto handling with the kernel defconfig....

bjharper, 2017-04-18:
Thanks for the quick response, with version 4.10 stable.  I might look at a different kernel config file and report back.

Cheers, Brad.

superna9999, 2017-04-18:
Hi,

Please try to rework I did on the branch 
https://github.com/superna9999/meta-meson/tree/krogoth-next

Neil

bjharper, 2017-04-18:
I've just tried krogoth-next with the same problem,  this is the Build configuration...

Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "aarch64-poky-linux"
MACHINE           = "hardkernel-odroidc2"
DISTRO            = "poky"
DISTRO_VERSION    = "2.0.3"
TUNE_FEATURES     = "aarch64"
TARGET_FPU        = ""
meta              
meta-yocto        
meta-yocto-bsp    = "jethro:e08994ce953ff5ab49b260e7763a40bfad4285ca"
meta-meson        = "krogoth-next:ebf06bf0fb26cc1973e503450394964374c3f6f7"

I will do some further debug, im thinking either my build environment has issues or the kernel config might be out a little and not displaying console.

Thanks,  Brad.

superna9999, 2017-04-18:
Hi,

Please update the krogoth-next branch to the lastest commit, I tested it on an Odroid-C2 board successfully.

And I also backported HDMI and Audio patches.

Neil

bjharper, 2017-04-18:
Hi Neil, I must be doing something wrong still seeing the same issue, I will look into things further.

I see these 2 warnings during build...

WARNING: Host distribution "Ubuntu-16.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.

And this one with the latest commit..
WARNING: defconfig detected in WORKDIR. defconfig skipped

Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "aarch64-poky-linux"
MACHINE           = "hardkernel-odroidc2"
DISTRO            = "poky"
DISTRO_VERSION    = "2.0.3"
TUNE_FEATURES     = "aarch64"
TARGET_FPU        = ""
meta              
meta-yocto        
meta-yocto-bsp    = "jethro:e08994ce953ff5ab49b260e7763a40bfad4285ca"
meta-meson        = "krogoth-next:14145ab0071afaba2e22d7e4e2d83652f52a160a"

-Brad

bjharper, 2017-04-18:
I will try following your new build example for the odroidc2 board exactly.

bjharper, 2017-04-18:
This time building the the complete image with wic create command and writing to sd I see the amlogic bootloader start but not uboot.  I tried writing to both an sdcard and a mmc card with the same results.

Previously I was using the amlogic-image-sata and was trying to refresh the meson-meta to keep the cache which may have trashed my environment.

Starting out clean again in a new amlogic-bsp directory and will report back, will take some time on my slow net connection / build machine so will report back tomorrow.

Thanks for all of your assistance today.

Best Regards,

Brad.

bjharper, 2017-04-19:
Ive pushed a little further using your latest commit (70ba2ceda6ad5c77eb242da95f64f63fcca9446f) but not without some issues.  Still struggling to get uboot to run with the build direct image using wic.

Hardkernel uboot environment (written using their default ubuntu image) will not boot the kernel with the uImage for me, always see nothing on the console when attempting to Starting Kernel....

Hardkernel uboot environment will not boot a standard Image file (one created by yocto build), ends in a Synchronous abort (think thier uboot has some custom environment variable which forces a modified entry point)

An upstream modified u-boot with mmc patches will boot the kernel in plain Image format (using the boot script below in the log) but kernel will not mount the rootfs.  I believe previously I have had to use a uinitrd rootfs to boot to achieve this with thier modified Ubuntu (using older ~4.9 linux-next version)

Boot log failing to mount rootfs...
--- UART initialized after reboot ---
[Reset cause: unknown]
[Image: unknown, amlogic_v1.1.3046-00db630 2015-10-28 15:24:31 xiaobo.gu@droid05]
bl30: check_permit, count is 1
bl30: check_permit: ok!
chipid: ef be ad de d f0 ad ba Load bl33 from SD, src: 0x00034200, des: 0x01000000, size: 0x00054f80
ef be ad de not ES chip
[0.487423 Inits done]
secure task start!
high task start!
low task start!
NOTICE:  BL3-1: v1.0(debug):4d2e34d
NOTICE:  BL3-1: Built : 17:08:35, Oct 29 2015
INFO:    BL3-1: Initializing runtime services
INFO:    BL3-1: Preparing for EL3 exit to normal world
INFO:    BL3-1: Next image address = 0x1000000
INFO:    BL3-1: Next image spsr = 0x3c9
No match for driver 'eth_designware'
Some drivers were not found

U-Boot 2016.07-rc2-gaf4187c (Jun 27 2016 - 11:39:05 +1000) odroid-c2

DRAM:  2 GiB
No maDMC:   <NULL>: 0
Using default environment

In:    serial@4c0
Out:   serial@4c0
Err:   serial@4c0
Net:   No ethernet found.
=> setenv kernel_addr 11000000
=> setenv fdt_addr 1000000
=> setenv bootfile Image
=> setenv fdtfile meson-gxbb-odroidc2.dtb
=> setenv fsargs "setenv bootargs root=/dev/mmcblk1p2 rw no_console_suspend fsck.fix=yes"
=> setenv bootcmd 'fatload mmc 0:1 ${initrd_loadaddr} ${initrdfile} ;fatload mmc 0:1 ${kernel_addr} ${bootfile};fatload mmc 0:1 ${fdt_addr} ${fdtfile};run fsargs ;booti ${kernel_addr} - ${fdt_addr}'
=> boot
dev_get_uclass_priv: null device
reading Image
14930432 bytes read in 4879 ms (2.9 MiB/s)
dev_get_uclass_priv: null device
reading Image
14930432 bytes read in 4880 ms (2.9 MiB/s)
dev_get_uclass_priv: null device
reading meson-gxbb-odroidc2.dtb
18650 bytes read in 12 ms (1.5 MiB/s)
## Flattened Device Tree blob at 01000000
   Booting using the fdt blob at 0x1000000
   Loading Device Tree to 000000007df53000, end 000000007df5a8d9 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.10.11-yocto-standard-meson64-meson (brad@ubuntu) (gcc version 5.2.0 (GCC) ) #1 SMP PREEMPT Wed Apr 19 03:52:34 AEST 2017
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000072000000, size 188 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[... 186 lines cut ...]
[    1.483471] Call trace:
[    1.485894] [<ffff000008088528>] dump_backtrace+0x0/0x238
[    1.491238] [<ffff000008088774>] show_stack+0x14/0x20
[    1.496239] [<ffff000008385bb4>] dump_stack+0x9c/0xc0
[    1.501242] [<ffff000008170824>] panic+0x11c/0x284
[    1.505985] [<ffff000008cd113c>] mount_block_root+0x188/0x264
[    1.511676] [<ffff000008cd1330>] mount_root+0x118/0x130
[    1.516852] [<ffff000008cd1484>] prepare_namespace+0x13c/0x184
[    1.522630] [<ffff000008cd0d78>] kernel_init_freeable+0x218/0x23c
[    1.528670] [<ffff0000088d8f90>] kernel_init+0x10/0x100
[    1.533844] [<ffff000008082ec0>] ret_from_fork+0x10/0x50
[    1.539104] SMP: stopping secondary CPUs
[    1.542985] Kernel Offset: disabled
[    1.546433] Memory Limit: none
[    1.549455] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

So it seems the hardkernel uboot will not display serial console (upstream uboot does) and the yocto kernel will not mount the rootfs and halt mid boot (at least without a uinitrd to load some modules)

I will look into the hardkernel u-boot to see if I can find anything starting here..
https://github.com/hardkernel/u-boot/blob/odroidc2-v2015.01/include/configs/odroidc2.h

I might be on the wrong track but I have a console for some debug now.  I will try a few more things tomorrow.

Thanks,
Brad.

superna9999, 2017-04-19:
Hi Brad,

Yes, there is two issues here:
- Hardkernel u-boot uses ttyS0 as serial console, because Amlogic kernel uses ttyS0
- Mainline u-boot uses the mainline linux ttyAML0

I was going to push a "mainline compatible" boot.ini, but you can find one here:
https://github.com/superna9999/LibreELEC.tv/blob/amlogic-gx/projects/Amlogic_GX/devices/Odroid_C2/bootloader/boot.ini

Second issue is that Amlogic changed the default "Image" address, so Hardkernel cannot boot mainline "Image", only uImage.

Keep uImage with Hardkernel U-boot but add the adapted boot.ini file in the FAT partition to have the correct console name !

Thanks for testing !!!!
Neil

bjharper, 2017-04-19:
Super its all making sense now!!

It late here now i will have to try tomorrow after work.  Thanks for all of your great development work.

Brad

bjharper, 2017-04-20:
Hi Neil,
    its now booting with a couple of modifications to the boot.ini you mentioned above.

I needed to add a kernel command line parameter for mmc..
root=/dev/mmcblk0p2

or for sd
root=/dev/mmcblk1p2

I also needed to remove the following LibreELEC option or for some reason it could not find root fs and I received a stack trace

setenv libreelec "quiet"

Working with both hdmi and serial console, only small thing I noticed was that I cannot use keyboard over the ttyAML0 if I also have usb keyboard connected to the C2 but c2's keyboard works fine over HDMI console

This was using the amlogic-image-headless-sd version of the build and manually installing boot and rootfs partitions.  Will now look at building up and testing amlogic-image-headless-sata.

Big Thanks!

superna9999, 2017-04-21:
Thanks for testing, I'll push a changed boot.ini and merge into main krogoth branch.

Neil

bjharper, 2017-04-21:
Great, lastly to enable wic to produce a full image which can be directly written to the sdcard and booted I think the hardekernel sd_fusing.sh script might need to be run against the final image to install uboot at the right location

Normally it takes the form $ sd_fusing.sh /dev/mmcblk0 but looks like the /dev/mmcblk0 can be replaced with the final wic generated image file.  Ill do some testing, I wonder if wic can achieve this itself.

closed 2017-04-26 by bjharper