superna9999/meta-meson #7: Usb and Sound for S912

150balbes, 2017-09-07:
Hi. 
I'm not much change the your patches (to support USB S912 and sound all S905X\S912) and added a new fee (s96 vega group s912). Added a script for wic, which collects the image to the external media with the specific launch with the use of universal multi-boot. If you're interested, it's all to be in my last commits on branches master and pyro. By the way, for some reason I can't create a request to merge in your GIT. You lock this or am I doing something wrong ?

superna9999, 2017-09-07:
Hi,
Thanks I wil have a look right now.

I didn't change anything, you should be able to create a pull request.

Maybe you can try to reset your branch to mine (on command line, fetch my git repo and checkout my branch, apple your fixes and push it to your git repo)

150balbes, 2017-09-07:
I'll see what I'm doing wrong.

closed 2018-04-23 by superna9999