superna9999/meta-meson #14: can't boot odroid-c2

twoerner, 2018-03-13:
Hi, I can't get an image based on this layer to boot on my odroid-c2 board following the instructions in the README file. Building an image with git://github.com/akuster/meta-odroid builds and boots fine on the exact same hardware (including SDcard).

I've tried rocko and master, OE/nodistro and poky. No combination was able to build successfully without https://github.com/superna9999/meta-meson/pull/13, but even after successfully building, none booted. I've tried core-image-minimal, amlogic-image-sato, and amlogic-image-headless-sd.

The boot process doesn't seem to be leaving the firmware, it starts with:

`GXBB:BL1:08dafd:0a8993;FEAT:EDFC318C;POC:3;RCY:0;EMMC:800;NAND:81;SD:0;READ:0;CHK:F3;USB:8;`

and then keeps repeating

`LOOP:1;EMMC:800;NAND:81;SD:0;READ:0;CHK:F3;USB:8;`

over and over.

After writing the SDcard with the meta-odroid image, the partitioning looks like:

```
# fdisk -l /dev/sdi
Disk /dev/sdi: 14.4 GiB, 15489564672 bytes, 30253056 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x71ec0000

Device     Boot  Start    End Sectors  Size Id Type
/dev/sdi1         2048 264191  262144  128M  e W95 FAT16 (LBA)
/dev/sdi2       264192 481279  217088  106M 83 Linux
```

after meta-meson:

```
# fdisk -l /dev/sdi
Disk /dev/sdi: 14.4 GiB, 15489564672 bytes, 30253056 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xe16276b6

Device     Boot Start     End Sectors   Size Id Type
/dev/sdi1  *        8   54763   54756  26.8M  c W95 FAT32 (LBA)
/dev/sdi2       54768 1141911 1087144 530.9M 83 Linux
```

The meta-odroid image also places a boot.scr file in the first (DOS) partition, which the meta-meson one doesn't (not sure if this is relevant).

superna9999, 2018-03-13:
Hi, 
Indeed there is a .scr missing in the boot partition, but you must also run the sd-fusing script installed in the deploy partition to install U-boot. It's not automatic yet.

twoerner, 2018-03-13:
Running the sd-fusing script on the SDcard (after dding the *mmcblk.direct wic output) seems to get further. It looks like bl1 and bl2 are running, but it keeps resetting over and over...

```
GXBB:BL1:08dafd:0a8993;FEAT:EDFC318C;POC:3;RCY:0;EMMC:800;NAND:81;SD:0;READ:0;CHK:0;
TE: 345181
no sdio debug board detected 

BL2 Built : 11:44:26, Nov 25 2015. 
gxb gfb13a3b-c2 - jcao@wonton

Board ID = 8
set vcck to 1100 mv
set vddee to 1050 mv
CPU clk: 1536MHz
DDR channel setting: DDR0 Rank0+1 same
DDR0: 2048MB(auto) @ 912MHz(2T)-13
DataBus test pass!
AddrBus test pass!
Load fip header from SD, src: 0x0000c200, des: 0x01400000, size: 0x000000b0
aml log : SIG CHK : 351 for address 0x01700000
reset...
```

I'm not sure if there's something I can do about the missing .scr

superna9999, 2018-03-14:
Hi, seems there is an issue with the generate U-Boot.

Can you try to follow these steps ?

twoerner, 2018-04-23:
Very nice! Thanks :-)