superna9999/meta-meson #20: topic/mesa-lima kernel hdmi issues

twoerner, 2018-04-23:
First off: I'm _very_ excited to see the topic/mesa-lima effort!!

It builds fine, but I haven't been able to see anything. I tried two different HDMI devices and both failed. The first is a [Vizio D32x-D1 TV](https://support.vizio.com/s/article/D32x-D1-Model-Information?language=en_US), and the second is a [Waveshare 1024x600 touchscreen USB Rev2.1](https://www.waveshare.com/wiki/7inch_HDMI_LCD_(C)).

With the Vizio: the 4 penguins show up on the monitor, but no text (i.e. there is no "Please wait, booting..."). As the kernel is booting, there is a long pause, then an error. It never gets to a prompt.

```
...snip...
[    2.066715] meson-drm d0100000.vpu: Queued 2 outputs on vpu
[    2.066888] meson-drm d0100000.vpu: Failed to create debugfs directory
[    2.073140] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    2.079673] [drm] No driver support for vblank timestamp query.
[    2.085590] meson-drm d0100000.vpu: CVBS Output connector not available
[    2.117812] meson-dw-hdmi c883a000.hdmi-tx: Detected HDMI TX controller v2.01a with HDCP (meson_dw_hdmi_phy)
[    2.122212] meson-dw-hdmi c883a000.hdmi-tx: registered DesignWare HDMI I2C bus driver
[    2.130023] meson-drm d0100000.vpu: bound c883a000.hdmi-tx (ops meson_dw_hdmi_ops)
[    2.454339] Console: switching to colour frame buffer device 240x67
[    2.487089] meson-drm d0100000.vpu: fb0:  frame buffer device
[   23.521770] INFO: rcu_preempt detected expedited stalls on CPUs/tasks: { 1-... } 5258 jiffies s: 9 root: 0x2/.
[   23.526101] blocking rcu_node structures:
[   23.530069] Task dump for CPU 1:
[   23.533256] irq/32-dw_hdmi_ R  running task        0   104      2 0x00000000
[   23.540247] Call trace:
[   23.542670]  __switch_to+0x94/0xd8
[   23.546031]  __dw_hdmi_setup_rx_sense+0x28/0x98
[   23.550512]  dw_hdmi_setup_rx_sense+0x10/0x18
[   23.554825]  dw_hdmi_top_thread_irq+0x2c/0x50
[   23.559137]  irq_thread_fn+0x28/0x68
[   23.562672]  irq_thread+0x10c/0x1a0
[   23.566126]  kthread+0x128/0x130
[   23.569309]  ret_from_fork+0x10/0x18
```

With the Waveshare, it's as though the device doesn't see it at all. It boots just fine all the way up to the prompt, but nothing ever comes up on the screen, and the X server fails to run, saying no screens can be found.

Build info:

```
Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "opensuse-42.3"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "hardkernel-odroidc2"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta-meson           = "topic/mesa-lima:8175b9906adbce166e36d47cb01a542a741f9013"
meta                 = "master:1f53140528d79c38d4f3a82cd0a03bd0ddc87275"
meta-oe              = "master:f0a664f2b7ac42fa8832ecc016ddf8b6260ee50e"
```

koenkooi, 2018-04-24:
Console is pointed to the serial port by u-boot, so you won't get text on screen, only when userspace launches a getty.

twoerner, 2018-04-24:
@koenkooi sorry I wasn't clear. Yes, the console scrolls on the serial port. But normally on the screen I'll see 4 penguins, the text "Please wait: booting...", then the OS banner: "OpenEmbedded nodistro.0 hardkernel-odroidc2 /dev/tty1", then the login prompt: "hardkernel-odroidc2 login:". What I'm pointing out in my report is that when the kernel crashes (or whatever it's doing) all I see are the 4 penguins, there is no "Please wait: booting..." text, and no text at all (but I'm not saying that I was expecting the console text).

twoerner, 2018-04-24:
As an update, I then tried the above on the libretech-cc board (using the exact same Vizio TV and touchpanel), and the kernel is able to boot without any hiccups every time. So it works on the libretech-cc board, but not on the odroid-c2 board.

NOTE: by saying it "works" all I'm saying is that the kernel doesn't hiccup, which is the topic of this issue report. In the case of the Vizio: the OpenEmbedded logo appears and the progress bar gets updated, however the X server fails to start. In the case of the touchscreen: during the boot I see the backlight come on, but no graphics ever appear on the screen, also the X server also fails to start.

koenkooi, 2018-04-24:
FWIW I think the Xorg issue isn't HDMI related, but to DRM bugs and Xorg needing a config:
```
[Tue Apr 24 15:34:58 2018] WARNING: CPU: 0 PID: 343 at /build/angstrom/v2017.12/build/tmp-angstrom-glibc/work-shared/libretech-cc/kernel-source/drivers/gpu/drm/drm_vblank.c:1015 drm_vblank_put+0xa8/0xc0
[Tue Apr 24 15:34:58 2018] Modules linked in: lima meson_rng gpu_sched meson_ir rng_core rc_core meson_gxbb_wdt ip_tables x_tables ipv6
[Tue Apr 24 15:34:58 2018] CPU: 0 PID: 343 Comm: htop Tainted: G        W        4.16.3 #7
[Tue Apr 24 15:34:58 2018] Hardware name: Libre Technology CC (DT)
[Tue Apr 24 15:34:58 2018] pstate: 20000085 (nzCv daIf -PAN -UAO)
[Tue Apr 24 15:34:58 2018] pc : drm_vblank_put+0xa8/0xc0
[Tue Apr 24 15:34:58 2018] lr : drm_crtc_vblank_put+0x14/0x20
[Tue Apr 24 15:34:58 2018] sp : ffff000008003e10
[Tue Apr 24 15:34:58 2018] x29: ffff000008003e10 x28: ffff80006be38000 
[Tue Apr 24 15:34:58 2018] x27: 0000000000000001 x26: ffff000008991878 
[Tue Apr 24 15:34:58 2018] x25: ffff000008b6e287 x24: ffff8000715aa800 
[Tue Apr 24 15:34:58 2018] x23: 000000000000001f x22: ffff000008003f0c 
[Tue Apr 24 15:34:58 2018] x21: 0000000000000080 x20: ffff800070fa1018 
[Tue Apr 24 15:34:58 2018] x19: ffff800071007418 x18: 0000ffffbe66c490 
[Tue Apr 24 15:34:58 2018] x17: 0000ffffbe64fa10 x16: 0000ffffbe67d2b8 
[Tue Apr 24 15:34:58 2018] x15: 0000ffffbe5d6c40 x14: 0000000000000000 
[Tue Apr 24 15:34:58 2018] x13: 0000000000000000 x12: 0000000000000000 
[Tue Apr 24 15:34:58 2018] x11: 0000fffff0b5d001 x10: 0000000000000040 
[Tue Apr 24 15:34:58 2018] x9 : ffff000008b12a30 x8 : 0000000040000000 
[Tue Apr 24 15:34:58 2018] x7 : 0000000000210d00 x6 : 0000000000000004 
[Tue Apr 24 15:34:58 2018] x5 : ffff00000848edc8 x4 : 0000000000000001 
[Tue Apr 24 15:34:58 2018] x3 : 0000000000000160 x2 : 0000000000000000 
[Tue Apr 24 15:34:58 2018] x1 : ffff800071027400 x0 : ffff8000713bc000 
[Tue Apr 24 15:34:58 2018] Call trace:
[Tue Apr 24 15:34:58 2018]  drm_vblank_put+0xa8/0xc0
[Tue Apr 24 15:34:58 2018]  drm_crtc_vblank_put+0x14/0x20
[Tue Apr 24 15:34:58 2018]  meson_crtc_irq+0x5c/0x160
[Tue Apr 24 15:34:58 2018]  meson_irq+0x20/0x30
[Tue Apr 24 15:34:58 2018]  __handle_irq_event_percpu+0x5c/0x148
[Tue Apr 24 15:34:58 2018]  handle_irq_event_percpu+0x1c/0x58
[Tue Apr 24 15:34:58 2018]  handle_irq_event+0x48/0x78
[Tue Apr 24 15:34:58 2018]  handle_fasteoi_irq+0xa8/0x180
[Tue Apr 24 15:34:58 2018]  generic_handle_irq+0x24/0x38
[Tue Apr 24 15:34:58 2018]  __handle_domain_irq+0x5c/0xb8
[Tue Apr 24 15:34:58 2018]  gic_handle_irq+0x58/0xa8
[Tue Apr 24 15:34:58 2018]  el0_irq_naked+0x4c/0x54
[Tue Apr 24 15:34:58 2018] ---[ end trace cd8c9ac1a66b714e ]---
```
```
[  4486.418] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[  4486.463] (II) modeset(G0): using drv /dev/dri/card0
[  4486.464] (EE) No devices detected. 
[  4486.464] (EE) 
Fatal server error:
[  4486.464] (EE) no screens found(EE) 
[  4486.464] (EE) 
````

twoerner, 2018-04-24:
Agreed. The only reason I brought it up was to clarify what I meant when I said "it works". It works insofar as the kernel is able to boot (i.e. this issue), but it doesn't work as far as actually bringing up the X server :-)

wrt a server config: I had been looking at http://git.yoctoproject.org/cgit/cgit.cgi/meta-freescale/tree/recipes-graphics/xorg-xserver/xserver-xf86-config/use-mainline-bsp/xorg.conf and trying to find inspiration, but so far no luck

twoerner, 2018-04-24:
btw I haven't seen the drm kernel issue you've posted. The only kernel dump I've seen is on the odroid-c2 related to hdmi_setup. I do see the same X server failure, however.

superna9999, 2018-04-27:
For the waveshare, it's normale since the 1024x600 is not a VESA standard mode and we do not support yet modes given by the EDID, but it's planned.

For the TV, there is no reason it does not work.
Can you connect with the UART console and dump the kernel dmesg ?

twoerner, 2018-04-27:
In my case (booting topic/mesa-lima on an odroid-c2) the kernel doesn't finish booting, therefore there is no cmdline (and hence, no ability to dump dmesg). All I see on my UART console is what I've posted in my initial report.

Booting topic/mesa-lima on a libretech-cc board succeeds, but then the X server fails to start (i.e. a different issue).

It seems as though the kernel boot problem I'm seeing (i.e. the topic of this issue) is related to odroid-c2 vs libretech-cc since this problem is only seen on the odroid-c2 and not the libretech-cc.

superna9999, 2018-09-16:
Should be fixed using linux 4.18 on `sumo-next`

closed 2018-09-16 by superna9999