superna9999/meta-meson #29: Finding correct image version

LeosSire, 2018-05-30:
Please forgive my ignorance, I come from an OpenWRT background so OE/Yocto is new to me.

I am interested is using the Amlogic S905x box (Best bang for buck) but how do i determine which of the S905x images I should use? 

I hope this isn't a stupid question, I don't want to brick the device first time out. 

Thank you,

LeosSire, 2018-05-30:
What a concise response, thank you,

[(https://github.com/kraj/meta-openwrt)] offers a very useful compilation routine covering the basics of OE which I am running through, its OpenWRT after all. Compiling the QEMUARM image to play with in the mean time.

I'm looking at using the Android TV Box as they are cheap-as-chips. 

Thank you,

twoerner, 2018-05-30:
The Yocto mailing list and IRC channel can be quite helpful:
https://www.yoctoproject.org/community/
https://www.yoctoproject.org/community/mailing-lists/