superna9999/meta-meson #30: Conflict in appropriate files - need help

LeosSire, 2018-06-02:
Hello,

I have been trying to compile my first image but I have come up against the below issue.

I have seen somewhere that I should pin a file, but how do I do this? I can't find a tutorial or any reference to what that means. 

`================================================================================
Install  612 Packages

Total size: 110 M
Installed size: 403 M
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Error: Transaction check error:
  file /etc/iproute2/rt_protos conflicts between attempted installs of base-files-3.0.14-r89.nexbox_a95x_s905x and iproute2-4.14.1-r0.aarch64
  file /etc/iproute2/rt_tables conflicts between attempted installs of base-files-3.0.14-r89.nexbox_a95x_s905x and iproute2-4.14.1-r0.aarch64
  file /etc/sysctl.conf        conflicts between attempted installs of base-files-3.0.14-r89.nexbox_a95x_s905x and procps-3.3.12-r0.aarch64

Error Summary
-------------

ERROR: amlogic-image-headless-sd-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/user/amlogic-bsp/build/tmp/work/nexbox_a95x_s905x-poky-linux/amlogic-image-headless-sd/1.0-r0/temp/log.do_rootfs.29264
ERROR: Task (/home/user/amlogic-bsp/meta-meson/recipes-core/images/amlogic-image-headless-sd.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3131 tasks of which 3130 didn't need to be rerun and 1 failed.
`

I have also getting the below warning. Is this related? I don't think it is. 

`WARNING: Layer meson should set LAYERSERIES_COMPAT_meson in its conf/layer.conf file to list the core layer names it is compatible with.`

Many thanks,

twoerner, 2018-06-07:
In order for anyone to help, you would have to provide more information.

Re-starting your work "fresh" (new shell, new location), recording the steps you did, then reporting them here would help. Also, at the start of every build, bitbake prints out a "Build Configuration" block; copying and pasting the whole block in here (including layer information) would help.

You'll probably have better luck with a build if all your layers are checked-out at the latest stable release: sumo. Can you visit each of the layers you're using, do a "git checkout sumo", and try again?

The warning is irrelevant.

superna9999, 2018-06-08:
We should fix this warning anyway !

LeosSire, 2018-06-08:
I managed to fix this with a tweak, put it in the busy-box tweaks. 

do_install_append() {
  rm -rf ${D}${sysconfdir}/iproute2/rt_tables
}

superna9999, 2018-06-09:
@leossire I don’t get how you have this error
Please be more explicit, or use the sumo branch without modifications with poky sumo and build a classic image like core-image-base

closed 2018-09-16 by superna9999