superna9999/meta-meson #37: Output files to single file

LeosSire, 2018-06-08:
Hello,

I have just compiled my first OE for nexbox_a95x_s905x which results in the below files.

There is no single flashable image. 

I understand the dtb is a hardware file, the ext4 is a mapped filesystem but not sure if this includes boot info. There is a separate boot image. 

The flashing script for my device requires a single image file. Is there a tool I can use to construct an img from the below files. 

Thank you, 

FILES:
aml_autoscript
amlogic-image-headless-sd-nexbox-a95x-s905x-20180608105022.rootfs.ext4
amlogic-image-headless-sd-nexbox-a95x-s905x-20180608105022.rootfs.manifest
amlogic-image-headless-sd-nexbox-a95x-s905x-20180608105022.rootfs.tar.bz2
amlogic-image-headless-sd-nexbox-a95x-s905x-20180608105022.testdata.json
amlogic-image-headless-sd-nexbox-a95x-s905x.ext4
amlogic-image-headless-sd-nexbox-a95x-s905x.manifest
amlogic-image-headless-sd-nexbox-a95x-s905x.tar.bz2
amlogic-image-headless-sd-nexbox-a95x-s905x.testdata.json
Image--4.16.5+git0+e5ce9f6879-r0-meson-gxl-s905x-nexbox-a95x-20180608105022.dtb
Image-meson-gxl-s905x-nexbox-a95x.dtb
meson-gxl-s905x-nexbox-a95x.dtb
modules--4.16.5+git0+e5ce9f6879-r0-nexbox-a95x-s905x-20180608105022.tgz
modules-nexbox-a95x-s905x.tgz
uImage
uImage--4.16.5+git0+e5ce9f6879-r0-nexbox-a95x-s905x-20180608105022.bin
uImage-nexbox-a95x-s905x.bin

LeosSire, 2018-06-08:
Getting there, i have to use the mkimage application but 

mkimage -A arm -O linux -T script -C none -d aml_autoscript.txt aml_autoscript

Doesn't work. It complains about a report a 'Missing output filename'.

Anyone assist?

superna9999, 2018-06-09:
Please be more explicit :
- which branch
- which oe/yocto release
- which image you build
- changes you did to the layer
- changes you did to the local.conf

closed 2018-09-16 by superna9999