superna9999/meta-meson #47: build failed for Khadas VIM

numbqq, 2018-06-30:
**Build Context**
 - meta-meson branch: **sumo**
 - meta-meson changes: **NO**
 - Yocto Project Poky / oe-core branch: **sumo**
 - local.conf changes: **NO**
 - bitbake command: **bitbake amlogic-image-sato**

**System (please complete the following information):**
 - Host OS: **Ubuntu**
 - Version: **16.04**
 - Architecture: **x86_64**

**Describe the bug**
When I try to build Yocto for Khadas VIM but failed with some warnings & errors.
Build command:
```
$ TEMPLATECONF=../meta-meson/conf
$ source poky/oe-init-build-env
$ export MACHINE=khadas-vim
$ bitbake amlogic-image-sato
```
But I got some warnings & errors:
```
WARNING: quilt-native-0.65-r0 do_fetch: Failed to fetch URL http://download.savannah.gnu.org/releases/quilt/quilt-0.65.tar.gz, attempting MIRRORS if available
WARNING: attr-native-2.4.47-r0 do_fetch: Failed to fetch URL http://download.savannah.gnu.org/releases/attr/attr-2.4.47.src.tar.gz, attempting MIRRORS if available
WARNING: popt-native-1.16-r3 do_fetch: Failed to fetch URL http://rpm5.org/files/popt/popt-1.16.tar.gz, attempting MIRRORS if available
WARNING: shadow-4.2.1-r0 do_fetch: Failed to fetch URL http://pkg-shadow.alioth.debian.org/releases/shadow-4.2.1.tar.xz, attempting MIRRORS if available
WARNING: libsndfile1-1.0.28-r0 do_fetch: Failed to fetch URL http://www.mega-nerd.com/libsndfile/files/libsndfile-1.0.28.tar.gz, attempting MIRRORS if available
WARNING: acl-2.2.52-r0 do_fetch: Failed to fetch URL http://download.savannah.gnu.org/releases/acl/acl-2.2.52.src.tar.gz, attempting MIRRORS if available
WARNING: iso-codes-3.77-r0 do_fetch: Failed to fetch URL https://pkg-isocodes.alioth.debian.org/downloads/iso-codes-3.77.tar.xz, attempting MIRRORS if available
WARNING: libsamplerate0-0.1.9-r1 do_fetch: Failed to fetch URL http://www.mega-nerd.com/SRC/libsamplerate-0.1.9.tar.gz, attempting MIRRORS if available
WARNING: lzip-native-1.19-r0 do_fetch: Failed to fetch URL http://download.savannah.gnu.org/releases/lzip/lzip-1.19.tar.gz, attempting MIRRORS if available
WARNING: sysvinit-2.88dsf-r14 do_fetch: Failed to fetch URL http://download.savannah.gnu.org/releases/sysvinit/sysvinit-2.88dsf.tar.bz2, attempting MIRRORS if available
WARNING: mktemp-1.7-r3 do_fetch: Failed to fetch URL ftp://ftp.mktemp.org/pub/mktemp/mktemp-1.7.tar.gz, attempting MIRRORS if available
WARNING: linux-yocto-meson64-4.17.0+gitAUTOINC+29dcea8877-r0 do_patch: 
Some of the context lines in patches were ignored. This can lead to incorrectly applied patches.
The context lines in the patches can be updated with devtool:

    devtool modify <recipe>
    devtool finish --force-patch-refresh <recipe> <layer_path>

Then the updated patches and the source tree (in devtool's workspace)
should be reviewed to make sure the patches apply in the correct place
and don't introduce duplicate lines (which can, and does happen
when some of the context is ignored). Further information:
http://lists.openembedded.org/pipermail/openembedded-core/2018-March/148675.html
https://bugzilla.yoctoproject.org/show_bug.cgi?id=10450
Details:
checking file arch/arm64/boot/dts/amlogic/meson-gxl.dtsi
Hunk #1 succeeded at 265 with fuzz 1.
WARNING: linux-yocto-meson64-4.17.0+gitAUTOINC+29dcea8877-r0 do_patch: 
Some of the context lines in patches were ignored. This can lead to incorrectly applied patches.
The context lines in the patches can be updated with devtool:

    devtool modify <recipe>
    devtool finish --force-patch-refresh <recipe> <layer_path>

Then the updated patches and the source tree (in devtool's workspace)
should be reviewed to make sure the patches apply in the correct place
and don't introduce duplicate lines (which can, and does happen
when some of the context is ignored). Further information:
http://lists.openembedded.org/pipermail/openembedded-core/2018-March/148675.html
https://bugzilla.yoctoproject.org/show_bug.cgi?id=10450
Details:
patching file arch/arm64/boot/dts/amlogic/meson-gxl.dtsi
Hunk #1 succeeded at 265 with fuzz 1.
ERROR: linux-yocto-meson64-4.17.0+gitAUTOINC+29dcea8877-r0 do_compile: oe_runmake failed
ERROR: linux-yocto-meson64-4.17.0+gitAUTOINC+29dcea8877-r0 do_compile: Function failed: do_compile (log file is located at /home/nick/code/git/amlogic/amlogic-bsp/build/tmp-glibc/work/khadas_vim-oe-linux/linux-yocto-meson64/4.17.0+gitAUTOINC+29dcea8877-r0/temp/log.do_compile.11487)
ERROR: Logfile of failure stored in: /home/nick/code/git/amlogic/amlogic-bsp/build/tmp-glibc/work/khadas_vim-oe-linux/linux-yocto-meson64/4.17.0+gitAUTOINC+29dcea8877-r0/temp/log.do_compile.11487
```
[... 242 lines cut ...]
|   CC      lib/crc-t10dif.o
|   CC      drivers/clk/clkdev.o
|   CC      fs/proc/task_mmu.o
|   CC      lib/crc-itu-t.o
|   CC      lib/crc7.o
|   CC      fs/nfs/io.o
|   CC      lib/genalloc.o
|   CC      drivers/clk/clk.o
|   CC      lib/percpu_counter.o
|   CC      lib/audit.o
|   CC      fs/nfs/direct.o
|   CC      lib/compat_audit.o
|   CC      fs/nfs/pagelist.o
|   CC      fs/proc/inode.o
|   CC      lib/swiotlb.o
|   CC      lib/syscall.o
|   CC      fs/proc/root.o
|   CC      lib/nlattr.o
|   CC      drivers/clk/clk-divider.o
|   CC      lib/checksum.o
|   CC      fs/nfs/read.o
|   CC      fs/proc/base.o
|   CC      lib/cpu_rmap.o

```

superna9999, 2018-06-30:
@numbqq Thanks for reporting, does it still fail on the master branch ?

superna9999, 2018-09-16:
@numbqq It should be fixed on `sumo, `sumo-next`and `master`branches

closed 2018-09-16 by superna9999