superna9999/meta-meson #48: u-boot-meson-gx build fails for hardkernel-odroidc2

reinhardd, 2018-07-04:
**Build Context**
 - meta-meson branch: master
 - meta-meson change: none
 - Poky branch: sumo
 - bitbake command: bitbake amlogic-image-sato
 - host: docker/stretch

The build
fails within do_deploy of u-boot-meson-gx:
```cp: cannot stat '/home/reinhard/Develop/odroid-c2/sumo/build/tmp-glibc/deploy/images/hardkernel-odroidc2/fip//bl31.img': No such file or directory```

Some debugging shows that do_deploy_append_hardkernel-odroidc2 does not get called. Instead do_deploy_append_meson-gxbb is used.

I get this fixed with:
```
diff --git a/conf/machine/hardkernel-odroidc2.conf b/conf/machine/hardkernel-odroidc2.conf
index 26172b9..17941c3 100644
--- a/conf/machine/hardkernel-odroidc2.conf
+++ b/conf/machine/hardkernel-odroidc2.conf
@@ -1,6 +1,11 @@
 # Hardkernel Odroid-C2 Configurations
 
-require conf/machine/include/amlogic-s905.inc
+# require conf/machine/include/amlogic-s905.inc
+
+require conf/machine/include/amlogic-meson64.inc
+
+MACHINE_FEATURES = "serial usbhost sdio screen wifi"
+
 require conf/machine/include/hardkernel-odroidc2-dtb.inc
 
 KERNEL_IMAGETYPE = "Image"
```
The resulting image boots successfully into GUI from sdcard.

f1vefour, 2018-07-10:
Seems you could create a pull request.

superna9999, 2018-07-10:
I’ll prefer fixing https://github.com/superna9999/meta-meson/blob/sumo/recipes-bsp/u-boot/u-boot-meson-gx_2018.01.bb instead

superna9999, 2018-07-10:
https://github.com/superna9999/meta-meson/blob/sumo/classes/image_types_meson.bbclass Uses an indirect WIC_COMMAND that seems to work better

olluz, 2018-08-28:
Where is this even coming from ? If I remove all occurrences of "bl31.img" in "u-boot-meson-gx_2018.01.bb" it will still fail to build:
images/hardkernel-odroidc2/fip//bl31.img': No such file or directory
I don't get it...

superna9999, 2018-09-16:
Fixed on `sumo`, `sumo-nex`and `master`

closed 2018-09-16 by superna9999