superna9999/meta-meson #49: No ethernet with Mecool M8S PRO+ (Using P212 dts)

intriguedlife, 2018-07-26:
I have successfully build and booted an amlogic-image-headless-sd image with the "amlogic-p212" machine on the Mecool M8S PRO+. The guys that are booting libreelec on this device are using the "gxl_p212_2g.dtb" file, thus it stands to reason that the "amlogic-p212" machine type is compatible.

Mainline u-boot was used so I had to erase the first sectors of the mmc as described in the README.

A problem that I encounter with this image is the not functioning ethernet.

I see no kernel errors:
> [    0.810343] meson8b-dwmac c9410000.ethernet: PTP uses main clock
[    0.815860] meson8b-dwmac c9410000.ethernet: no reset control found
[    0.827072] meson8b-dwmac c9410000.ethernet: Ring mode enabled
[    0.832846] meson8b-dwmac c9410000.ethernet: DMA HW capability register supported
[    0.840260] meson8b-dwmac c9410000.ethernet: Normal descriptors
[    0.846123] meson8b-dwmac c9410000.ethernet: RX Checksum Offload Engine supported
[    0.853541] meson8b-dwmac c9410000.ethernet: COE Type 2
[    0.858716] meson8b-dwmac c9410000.ethernet: TX Checksum insertion supported
[    0.865701] meson8b-dwmac c9410000.ethernet: Wake-Up On Lan supported
[    0.872116] meson8b-dwmac c9410000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[  123.285420] meson8b-dwmac c9410000.ethernet eth0: device MAC address 1e:84:7b:c8:84:92
[  123.387653] meson8b-dwmac c9410000.ethernet eth0: PTP not supported by HW

ethtool reports that there is no link detected even though I have a cable inserted:

> 
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Speed: Unknown!
        Duplex: Unknown! (255)
        Port: MII
        PHYAD: 8
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: ug
        Wake-on: d
        Current message level: 0x0000003f (63)
                               drv probe link timer ifdown ifup
        Link detected: no

I do see a couple of differences when I decompile the "gxl_p212_2g.dtb" file used by libreelec and the dtb file that is generated by my build using the "amlogic-p212" machine:

gxl_p212_2g.dtb:
> ethernet@0xc9410000 {
        compatible = "amlogic, gxbb-rmii-dwmac";
        reg = <0x00000000 0xc9410000 0x00000000 0x00010000 0x00000000 0xc8834540 0x00000000 0x00000008 0x00000000 0xc8834558 0x00000000 0x0000000c>;
        interrupts = <0x00000000 0x00000008 0x00000001 0x00000000 0x00000009 0x00000001>;
        phy-mode = "rmii";
        pinctrl-names = "internal_eth_pins";
        pinctrl-0 = <0x0000001e>;
        mc_val = <0x00001800>;
        resets = <0x00000008 0x00000023>;
        reset-names = "ethpower";
        interrupt-names = "macirq", "phyirq";
        clocks = <0x00000008 0x0000000a>;
        clock-names = "ethclk81";
        internal_phy = <0x00000001>;
        wol = <0x00000000>;
    };

amlogic-p212 (meson-glx-s905x-p212.dtb):

> ethernet@c9410000 {
            compatible = "amlogic,meson-gx-dwmac", "amlogic,meson-gxbb-dwmac", "snps,dwmac";
            reg = <0x00000000 0xc9410000 0x00000000 0x00010000 0x00000000 0xc8834540 0x00000000 0x00000004>;
            interrupts = <0x00000000 0x00000008 0x00000001>;
            interrupt-names = "macirq";
            status = "okay";
            clocks = <0x0000000b 0x00000024 0x0000000b 0x00000004 0x0000000b 0x0000000f>;
            clock-names = "stmmaceth", "clkin0", "clkin1";
            phy-mode = "rmii";
            phy-handle = <0x0000001b>;
            mdio {
                #address-cells = <0x00000001>;
                #size-cells = <0x00000000>;
                compatible = "snps,dwmac-mdio";
                phandle = <0x0000001a>;
            };
        };

The mainline kernel is obviously different from the Amlogic kernel so the problem might not lay here.

If anyone could assist me in getting the ethernet working?

superna9999, 2018-07-28:
Hi,

It looks like the Ethernet phy driver hasn’t probed, can you run a lsmod and share a full boot log (dmesg) ?

intriguedlife, 2018-07-28:
lsmod:

| le | Size | Used by |
| -- | ---- | -------- |
ipv6					|	385024 | 30
meson_ir				|	16384 | 0
rc_core				|	36864 | 2 meson_ir
meson_rng			|	16384 | 0
rng_core				|	16384 | 1 meson_rng
meson_gxbb_wdt		|	16384 | 0
pwm_meson			|	16384 | 1

dmesg:

> [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 4.17.1 (oe-user@oe-host) (gcc version 7.3.0 (GCC)) #1 SMP PREEMPT Fri Jul 27 13:57:22 UTC 2018
[    0.000000] Machine model: Amlogic Meson GXL (S905X) P212 Development Board
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000072000000, size 188 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 518912
[    0.000000]   DMA32 zone: 8128 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 518912 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] percpu: Embedded 23 pages/cpu @        (ptrval) s54552 r8192 d31464 u94208
[    0.000000] pcpu-alloc: s54552 r8192 d31464 u94208 alloc=23*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 510784
[    0.000000] Kernel command line: root=/dev/mmcblk0p1 rootwait rw console=ttyAML0
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Memory: 1835076K/2075648K available (7486K kernel code, 584K rwdata, 2716K rodata, 448K init, 343K bss, 48060K reserved, 192512K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000]  RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=4.
[    0.000000]  Tasks RCU enabled.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_meson_gpio: 110 to 8 gpio interrupt mux initialized
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000003] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000285] Console: colour dummy device 80x25
[    0.000304] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.000314] pid_max: default: 32768 minimum: 301
[    0.000381] Security Framework initialized
[    0.000418] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000431] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.020007] ASID allocator initialised with 32768 entries
[    0.023999] Hierarchical SRCU implementation.
[    0.032019] smp: Bringing up secondary CPUs ...
[    0.052183] Detected VIPT I-cache on CPU1
[    0.052228] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.072209] Detected VIPT I-cache on CPU2
[    0.072242] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.092242] Detected VIPT I-cache on CPU3
[    0.092274] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.092335] smp: Brought up 1 node, 4 CPUs
[    0.092354] SMP: Total of 4 processors activated.
[    0.092360] CPU features: detected: 32-bit EL0 Support
[    0.095128] CPU: All CPU(s) started at EL2
[    0.095141] alternatives: patching kernel code
[    0.095683] devtmpfs: initialized
[... 144 lines cut ...]
[    2.579649] random: udevd: uninitialized urandom read (16 bytes read)
[    2.582210] random: udevd: uninitialized urandom read (16 bytes read)
[    2.586889] random: udevd: uninitialized urandom read (16 bytes read)
[    2.634465] udevd[165]: starting eudev-3.2.5
[    2.699294] meson-gx-mmc d0070000.mmc: allocated mmc-pwrseq
[    2.715439] Registered IR keymap rc-empty
[    2.715540] rc rc0: meson-ir as /devices/platform/soc/c8100000.bus/c8100580.ir/rc/rc0
[    2.722414] input: meson-ir as /devices/platform/soc/c8100000.bus/c8100580.ir/rc/rc0/input0
[    2.732433] meson-ir c8100580.ir: receiver initialized
[    2.749090] mmc2: new high speed SDIO card at address 0001
[   62.972026] udevd[165]: worker [180] /devices/platform/soc/d0000000.apb/d0074000.mmc/mmc_host/mmc1/mmc1:0001/block/mmcblk1/mmcblk1boot0 is taking a long time
[   62.980785] udevd[165]: worker [179] /devices/platform/soc/d0000000.apb/d0074000.mmc/mmc_host/mmc1/mmc1:0001/block/mmcblk1/mmcblk1boot1 is taking a long time
[  122.873918] EXT4-fs (mmcblk0p1): re-mounted. Opts: (null)
[  123.408391] urandom_read: 3 callbacks suppressed
[  123.408403] random: dd: uninitialized urandom read (512 bytes read)
[  123.573779] meson8b-dwmac c9410000.ethernet eth0: device MAC address 1e:84:7b:c8:84:92
[  123.653204] Meson GXL Internal PHY 0.e40908ff:08: attached PHY driver [Meson GXL Internal PHY] (mii_bus:phy_addr=0.e40908ff:08, irq=37)
[  123.671388] meson8b-dwmac c9410000.ethernet eth0: PTP not supported by HW
[  132.998350] random: dbus-daemon: uninitialized urandom read (12 bytes read)
[  133.009468] random: dbus-daemon: uninitialized urandom read (12 bytes read)
[  133.264489] random: sshd: uninitialized urandom read (32 bytes read)
[  133.341799] NET: Registered protocol family 10
[  133.342860] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[  133.346712] Segment Routing with IPv6
[  184.262682] random: crng init done

intriguedlife, 2018-08-03:
It does seem to detect the phy, although the irq number looks wrong to me. When looking at dmesg logs of others the irq number is often reported as -1 for the internal phy.

This is what I get:
> [ 123.653204] Meson GXL Internal PHY 0.e40908ff:08: attached PHY driver [Meson GXL Internal PHY] (mii_bus:phy_addr=0.e40908ff:08, irq=37)

I have added some printk's in the stmmac driver when trying to debug. This is part of the dmesg from that debug:
> [    2.269594] meson8b-dwmac c9410000.ethernet: PTP uses main clock
[    2.270116] meson-ir c8100580.ir: receiver initialized
[    2.277143] meson8b-dwmac c9410000.ethernet: no reset control found
[    2.288167] stmmac - user ID: 0x11, Synopsys ID: 0x37
[    2.291823] meson8b-dwmac c9410000.ethernet: Ring mode enabled
[    2.297568] meson8b-dwmac c9410000.ethernet: DMA HW capability register supported
[    2.305005] meson8b-dwmac c9410000.ethernet: Normal descriptors
[    2.310907] meson8b-dwmac c9410000.ethernet: RX Checksum Offload Engine supported
[    2.318331] meson8b-dwmac c9410000.ethernet: COE Type 2
[    2.323507] meson8b-dwmac c9410000.ethernet: TX Checksum insertion supported
[    2.330483] meson8b-dwmac c9410000.ethernet: Wake-Up On Lan supported
[    2.336894] meson8b-dwmac c9410000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    2.344929] libphy: stmmac: probed
[    2.349005] libphy: mdio_mux: probed
[    2.364768] mmc2: new high speed SDIO card at address 0001
[    2.380992] libphy: mdio_mux: probed
[    2.443352] EXT4-fs (mmcblk0p1): re-mounted. Opts: (null)
[    3.171161] meson8b-dwmac c9410000.ethernet eth0: device MAC address 7e:87:3c:81:4f:61
[    3.232289] mdio read addr: 8 , 0
[    3.232529] mdio read finished data:
[    3.232644] 4096
[    3.248036] mdio read addr: 8 , 1
[    3.248408] mdio read finished data:
[    3.248509] 30729
[    3.251134] mdio read addr: 8 , 0
[    3.253321] mdio read finished data:
[    3.256456] 4096
[    3.260428] mdio read addr: 8 , 29
[    3.261771] mdio read finished data:
[    3.265042] 0
[    3.268744] Meson GXL Internal PHY 0.e40908ff:08: attached PHY driver [Meson GXL Internal PHY] (mii_bus:phy_addr=0.e40908ff:08, irq=37)
[    3.294756] meson8b-dwmac c9410000.ethernet eth0: PTP not supported by HW
[    3.296459] mdio read addr: 8 , 4
[    3.296664] mdio read finished data:
[    3.299246] 481
[    3.302900] mdio read addr: 8 , 1
[    3.306643] mdio read finished data:
[    3.307768] 30729
[    3.311332] mdio read addr: 8 , 0
[    3.314330] mdio read finished data:
[    3.316561] 4096
[    3.320027] mdio read addr: 8 , 1
[    3.321975] mdio read finished data:
[    3.325136] 30729

how to read the debug printk's: 
In the example below is 8 the address (This is correct and in accordance with dtb-bindings) and 0 is the register being read. 4096 is the integer value that was read at the register.
mdio read addr: 8 , 0
mdio read finished data:
4096

@superna9999 Can you confirm if these mdio read results are correct?

intriguedlife, 2018-08-11:
I may have found where the problem is. I think mainline u-boot on this board does not set the ethernet phy up properly.

In the beginning I was unable to get a mainline kernel booting with the vendor u-boot that was installed on the internal mmc of the device. That's why I started working with a mainline u-boot from the start.

But trying to solve the not working ethernet got me to try a lot of different things. Just now I discovered that when I build an amlogic u-boot(that did happen to boot a mainline kernel) and wrote that to the sdcard(overwriting mainline u-boot) with mainline kernel on it that ethernet did work.

superna9999, 2018-08-14:
Hi,

We usually don't change the U-boot from actual products, only from SBCs and reference designs since we have the source code.

We may be missing some init in mainline u-boot, we should investigate.

intriguedlife, 2018-08-28:
I have found the bug in mainline u-boot that prevents the phy from working with an p212 board.

The following code only gets executed if "flags" is not zero:
https://github.com/u-boot/u-boot/blob/11ed312896c5f5814064c5d45dcb2f53dc121437/arch/arm/mach-meson/eth.c#L37-L42

However in the board init code flags is set to zero:
https://github.com/u-boot/u-boot/blob/11ed312896c5f5814064c5d45dcb2f53dc121437/board/amlogic/p212/p212.c#L32

Therefore the code is not executed and the internal GXL RMII PHY is not working.

superna9999, 2018-08-28:
Hi, you are right it’s a stupid error of mine... it should be fixed upstream and backported to meta-meson. Do you know how to submit a fix to uboot upstream ?

intriguedlife, 2018-08-28:
I have not submitted a patch to u-boot before but I would like to try.

Should the patch add the change for odroid-c2 and nanopi-k2 as well? These are s905 boards with internal gigabit phy's so I don't know if they need this change as well.

superna9999, 2018-08-28:
Exact, The fix is only for p212, odroid-c2 and nanopi k2 are s905 SoC without internal phy and they use an external rgmii phy

intriguedlife, 2018-09-11:
Fixed with d8d20768de296abae33da5f4c061b504bcf36f7a

closed 2018-09-11 by intriguedlife