superna9999/meta-meson #53: Cant compile. Stuck on do.rootfs

ffnull, 2018-09-06:
**Describe the bug**

NOTE: Executing RunQueue Tasks
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: busybox.postinst returned 1, marki                                                        ng as unpacked only, configuration required on target.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: Intentionally failing postinstall                                                         scriptlets of ['busybox'] to defer them to first boot is deprecated. Please plac                                                        e them into pkg_postinst_ontarget_${PN} ().
If deferring to first boot wasn't the intent, then scriptlet failure may mean an                                                         issue in the recipe, or a regression elsewhere.
Details of the failure are in /home/ffnull/amlogic-bsp/build/tmp-glibc/work/frie                                                        ndlyelec_nanopik2-oe-linux/amlogic-image-sato/1.0-r0/temp/log.do_rootfs.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: busybox.postinst returned 1, marki                                                        ng as unpacked only, configuration required on target.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: Intentionally failing postinstall                                                         scriptlets of ['busybox'] to defer them to first boot is deprecated. Please plac                                                        e them into pkg_postinst_ontarget_${PN} ().
If deferring to first boot wasn't the intent, then scriptlet failure may mean an                                                         issue in the recipe, or a regression elsewhere.
Details of the failure are in /home/ffnull/amlogic-bsp/build/tmp-glibc/work/frie                                                        ndlyelec_nanopik2-oe-linux/amlogic-image-sato/1.0-r0/temp/log.do_rootfs.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: [log_check] amlogic-image-sato: fo                                                        und 2 warning messages in the logfile:
[log_check] WARNING: Intentionally failing postinstall scriptlets of ['busybox']                                                         to defer them to first boot is deprecated. Please place them into pkg_postinst_                                                        ontarget_${PN} ().
[log_check] WARNING: Intentionally failing postinstall scriptlets of ['busybox']                                                         to defer them to first boot is deprecated. Please place them into pkg_postinst_                                                        ontarget_${PN} ().

update-alternatives: Error: not linking /home/ffnull/amlogic-bsp/build/tmp-glibc/work/friendlyelec_nanopik2-oe-linux/amlogic-image-sato/1.0-r0/rootfs/sbin/klogd to /bin/busybox.nosuid since /home/ffnull/amlogic-bsp/build/tmp-glibc/work/friendlyelec_nanopik2-oe-linux/amlogic-image-sato/1.0-r0/rootfs/sbin/klogd exists and is not a link

**Build Context**
 - meta-meson branch
amlogic-image-sato
 - meta-meson changes
no
 - Yocto Project Poky / oe-core branch
sumo
 - other layers enabled in bblayers and their priorities
no
 - local.conf changes
no
 - bitbake command
bitbake amlogic-image-sato

**Expected behavior**
A clear and concise description of what you expected to happen.

**System (please complete the following information):**
 - Host OS: VBOX Ubuntu 16.04 X64

ffnull, 2018-09-08:
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: busybox.postinst returned 1, marking as unpacked only, configuration required on target.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: Intentionally failing postinstall scriptlets of ['busybox'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().
If deferring to first boot wasn't the intent, then scriptlet failure may mean an issue in the recipe, or a regression elsewhere.
Details of the failure are in /home/ffnull/amlogic-bsp/build/tmp-glibc/work/friendlyelec_nanopik2-oe-linux/amlogic-image-sato/1.0-r0/temp/log.do_rootfs.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: busybox.postinst returned 1, marking as unpacked only, configuration required on target.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: Intentionally failing postinstall scriptlets of ['busybox'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().
If deferring to first boot wasn't the intent, then scriptlet failure may mean an issue in the recipe, or a regression elsewhere.
Details of the failure are in /home/ffnull/amlogic-bsp/build/tmp-glibc/work/friendlyelec_nanopik2-oe-linux/amlogic-image-sato/1.0-r0/temp/log.do_rootfs.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: [log_check] amlogic-image-sato: found 2 warning messages in the logfile:
[log_check] WARNING: Intentionally failing postinstall scriptlets of ['busybox'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().
[log_check] WARNING: Intentionally failing postinstall scriptlets of ['busybox'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().

superna9999, 2018-09-16:
Hi, Sorry I don't understand how you get these messages.

Can you precise on which branch of `meta-meson` you are ?

ffnull, 2018-09-17:
sumo

superna9999, 2018-09-17:
I did the same steps multiple times drom a frest checkout, you should not have these warnings.

Can you check you have all the necessary packages installer on your distro ?

ffnull, 2018-09-17:
im using ubuntu 16.04 can you provide what i need to install?

superna9999, 2018-09-17:
https://www.yoctoproject.org/docs/2.5.1/ref-manual/ref-manual.html#required-packages-for-the-host-development-system

ffnull, 2018-09-17:
i will try again and will write you

ffnull, 2018-09-18:
Fresh new ubuntu. Still same

`WARNING: mobile-broadband-provider-info-1_20170310-r0 do_fetch: Failed to fetch URL git://git.gnome.org/mobile-broadband-provider-info, attempting MIRRORS if available
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: busybox.postinst returned 1, marking as unpacked only, configuration required on target.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: Intentionally failing postinstall scriptlets of ['busybox'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().
If deferring to first boot wasn't the intent, then scriptlet failure may mean an issue in the recipe, or a regression elsewhere.
Details of the failure are in /home/ffnull/amlogic-bsp/build/tmp-glibc/work/friendlyelec_nanopik2-oe-linux/amlogic-image-sato/1.0-r0/temp/log.do_rootfs.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: busybox.postinst returned 1, marking as unpacked only, configuration required on target.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: Intentionally failing postinstall scriptlets of ['busybox'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().
If deferring to first boot wasn't the intent, then scriptlet failure may mean an issue in the recipe, or a regression elsewhere.
Details of the failure are in /home/ffnull/amlogic-bsp/build/tmp-glibc/work/friendlyelec_nanopik2-oe-linux/amlogic-image-sato/1.0-r0/temp/log.do_rootfs.
WARNING: amlogic-image-sato-1.0-r0 do_rootfs: [log_check] amlogic-image-sato: found 2 warning messages in the logfile:
[log_check] WARNING: Intentionally failing postinstall scriptlets of ['busybox'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().
[log_check] WARNING: Intentionally failing postinstall scriptlets of ['busybox'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().

`

Detailed log https://pastebin.com/TYy1Xt2X

superna9999, 2018-09-19:
Hi,

The do_rootfs seems to have finished, what's the problem ?

ffnull, 2018-09-19:
its not finished correctly. so i cant passed login. no login and password

superna9999, 2018-09-19:
Can you try to build a poky only platform, like beaglebone-yocto and core-image-base without meta-meson ?

intriguedlife, 2018-09-23:
I have also gotten these postinst warnings for busybox every time I compiled.
In order to login I had to create build/conf/local.conf with the following:

> EXTRA_IMAGE_FEATURES = "debug-tweaks"

This lets you login as root without a password.

superna9999, 2018-09-26:
@ffnull ok, I know, please copy the file `poky/meta-poky/conf/local.conf.sample` into `build/conf/local.conf' and run bitbake again.

I will update the README.

superna9999, 2018-11-17:
Add earlycon to the boot params and share the boot log, thanks

Greg798100, 2018-11-17:
earlycon - It is difficult for me, sorry
here screenshot
[https://ibb.co/nNixx0](url)

Greg798100, 2018-11-17:
superna9999
Well?
Who is your code for?
For yourself?
Or are you so great to descend?
Maybe you are God?

superna9999, 2018-11-17:
Hi, sorry for not replying faster but I have some family duties, and this project is community driven and I develop it on my spare time.
This yocto layer doesn’t any weird stuff and you should be able to debug your issue with classic embedded Linux knowledge and using the huge Yocto documentation. You can also go on the freenode #linux-amlogic irc channel or Linux-amlogic mailing list as described on http://Linux-meson.com

I’ll have a look at your issue ASAP, but I can’t right now.

Greg798100, 2018-11-18:
Thank you.
I did according to your amlogic q201 manual and regret getting these errors.
It was in raspberry and corrected, but here it did not help.
https://raspberrypi.stackexchange.com/questions/40854/kernel-panic-not-syncing-vfs-unable-to-mount-root-fs-on-unknown-block179-6

Greg798100, 2018-11-18:
> Changing 200000000 to 100000000 on this line:
> https://github.com/torvalds/linux/blob/1ce80e0fe98e72af4125f47c65b0bc01b60a7feb/arch/arm64/boot/dts/amlogic/meson-gx-p23x-q20x.dtsi#L205
> 
> That may help. My rootfs also wouldn't mount unless I changed the max-frequency from 200000000 to 100000000. I had my rootfs on the internal mmc storage though.

Thanks
Your up 200000000 sd card?
There are cases damaged.

Greg798100, 2018-11-18:
Im changed for my sd 200000000 and kernel panic...
 Im see script boot balbes 150, maby this fix?
`'^E^YV��f)[+��^@^@^D�^@^@^@^@^@^@^@^@��p�^E^V^F^@s905_autoscript^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^D�^@^@^@^@setenv kernel_loadaddr "0x11000000"
setenv initrd_loadaddr "0x13000000"
setenv dtb_mem_addr "0x1000000"
setenv bootargs "console=ttyAML0,115200n8 console=tty0 no_console_suspend consoleblank=0 root=/dev/mmcblk0p2 rootwait"
if fatload mmc 0 ${kernel_loadaddr} uImage; then if fatload mmc 0 ${dtb_mem_addr} dtb.img ; then bootm ${kernel_loadaddr} - ${dtb_mem_addr} ; fi ; fi
setenv bootargs "console=ttyAML0,115200n8 console=tty0 no_console_suspend consoleblank=0 root=/dev/mmcblk1p2 rootwait"
if fatload mmc 1 ${kernel_loadaddr} uImage; then if fatload mmc 1 ${dtb_mem_addr} dtb.img ; then bootm ${kernel_loadaddr} - ${dtb_mem_addr} ; fi ; fi
setenv bootargs "console=ttyAML0,115200n8 console=tty0 no_console_suspend consoleblank=0 root=/dev/sda2 rootwait"
if fatload usb 0 ${kernel_loadaddr} uImage; then if fatload usb 0 ${dtb_mem_addr} dtb.img ; then bootm ${kernel_loadaddr} - ${dtb_mem_addr} ; fi ; fi
setenv bootargs "console=ttyAML0,115200n8 console=tty0 no_console_suspend consoleblank=0 root=/dev/sdb2 rootwait"
if fatload usb 1 ${kernel_loadaddr} uImage; then if fatload usb 1 ${dtb_mem_addr} dtb.img ; then bootm ${kernel_loadaddr} - ${dtb_mem_addr} ; fi ; fi
`

Greg798100, 2018-11-20:
Thanks all.
Strange, rebuilt on amlogic-s9xxx, and boot!
Fix my kernel panic cannot open root device.

closed 2019-01-23 by superna9999