superna9999/meta-meson #54: Add recipe for s905w

damdinsharav, 2018-09-13:
Hello,

Are you planning to add recipe for s905w based devices.

If you need dtb here it is. https://github.com/AlexELEC/S905-trees/blob/master/S905W/gxl_p281_1g.dtb

Let me know if you need anything else.

superna9999, 2018-09-13:
Hi @damdinsharav,

The support for P281 on s905w will land on Linux 4.19, when Linux 4.19 is tagged I will push support for it.

damdinsharav, 2018-09-26:
Hello @superna9999 ,

On http://linux-meson.com/doku.php, I see Amlogic P281 (GXL, S905W) and
Oranth Tanix TX3 Mini (GXL, S905W) are supported. 

Is there plan for this device? https://www.cnx-software.com/2017/08/24/x96-mini-amlogic-s905w-android-tv-box-sells-for-25-and-up/

superna9999, 2018-09-26:
@damdinsharav Hi, yes, when 4.19 is tagged.

closed 2018-09-26 by superna9999

damdinsharav, 2018-10-30:
Hello, now that kernel 4.19 released when can we expect it? 
Can't wait to try on my device. Thanks

damdinsharav, 2018-12-15:
Hello @superna9999 , do you have any update? Thanks