superna9999/meta-meson #55: Getting .py errors during the build process

Tommy21st, 2018-09-17:
I am getting .py errors during the build process, i made a terminal screenshot, am i maybe missing some python package, or maybe i need to edit some file.

I am on Ubuntu 16.04., making image for Vim1.

superna9999, 2018-09-17:
@Tommy21st please check https://www.yoctoproject.org/docs/2.5.1/ref-manual/ref-manual.html#required-packages-for-the-host-development-system before

Tommy21st, 2018-09-18:
I am using wubi for Ubuntu xenial installation, i even did a reinstall of the system, starting all over again.

I did follow yocto documentation, installed everything that is needed for Ubuntu, now i have even more errors, i think there are some packages missing from the documents, i have do compile or do configure errors, really can't understand what is wrong, i did have more than 10 logs in amlogic bsp folder when i've had those .py errors.

superna9999, 2018-09-19:
Can you try to build a poky only platform, like beaglebone-yocto and core-image-base without meta-meson ?

Tommy21st, 2018-09-19:
file:///home/khadas/Desktop/log.do_compile

I've now removed 16.04, i am using bionic from now, now this log turns up, can you read anything from it.

superna9999, 2018-09-21:
The khadas-vim-sdboot target is broken, please use the khadas-vim target instead which uses a more recent u-boot version.

Tommy21st, 2018-09-22:
I have managed to build the image, now i have problems of making it as a bootable image, i have used khadas-vim  target as you have said, i am using a dd like written in the readme file.

sudo dd if=/home/khadas/amlogic-bsp/build/tmp-glibc/deploy/images/khadas-vim.wic of=/dev/sdb1

dd: failed to open '/home/khadas/home/khadas/amlogic-bsp/build/tmp-glibc/deploy/images/khadas-vim.wic': No such file or directory

sdb1 is my sd card, sdb is a small 16mb partition on sd(automatically generated i guess), big one is sdb1.

The path to /khadas-vim.wic was copied using file manager path, i see the output is '/home/khadas/home/khadas, so this must be the cause, it repeats twice.

I've tried even this command: sudo dd if=~/home-khadas-amlogic-bsp-build-tmp-glibc-deploy-images-khadas-vim.wic of=/dev/sdb1, result is the same message.

Can you help?

Tommy21st, 2018-09-25:
Image done, one picture is from file manager, second one is from terminal, what is this, nothing is written on sd?
[image]
[image]

superna9999, 2018-09-26:
Hi,

You need to dd into /dev/sdb not /dev/sdb1, the .wic file contains the partitionning and /dev/sdb1 is the block file for the first partition of /dev/sdb

Tommy21st, 2018-09-26:
I tried that, but i didn't believe it takes so much time to make image, so i gave up the first time, i believe i waited an hour at least.

Now the image is ready, but it won't boot, i don't have this problem with Armbian.

I am using amlogic s905x image. I see 2 partitions in file manager, one is bootfs, other one is root fs. Bootfs has uimage and 4 or 5 dtb's, khadas vim included.

Do i need to change anything to boot it, with armbian it's simple, copying the device dtb from dtb folder, and renaming it to dtb.img, but i don't know how to boot this image?

Anyway, if you know or not know the cause of this issue, thank you for helping me! :)

superna9999, 2018-09-27:
Using the "khadas-vim" machine, you should only have a single ext4 root partition if you flash the .wic file on the /dev/sdb block device. It will work *only* if you erase the original u-boot on the eMMC.

If you want to boot using the vendor u-boot on eMMC, you will need to build for the khadas-vim-sdboot target. This image will hae 2 partitions, onle containing the s905_autoscript allowing to boot.

Please use a tool like https://etcher.io/ to flash the sdcard.

closed 2019-01-23 by superna9999