superna9999/meta-meson #58: eMMC error on Khadas VIM2

numbqq, 2019-01-23:
Hello Neil,

I use the patches from `https://github.com/superna9999/meta-meson/tree/master/recipes-kernel/linux/linux-yocto-meson64-4.20`,  and run linux 4.20.3 on Khadas VIM2, but I found the eMMC not work?

```
root@Khadas:~# dmesg | grep mmc1
[    4.550534] mmc1: error -110 whilst initialising MMC card
```

Do you have this issue?

Thanks.

superna9999, 2019-01-28:
Can you try removing:
```
mmc-hs400-1_8v;
```
and
```
mmc-hs200-1_8v;
```
?

numbqq, 2019-02-14:
Hello @superna9999 ,

When I remove `mmc-hs400-1_8v;` and `mmc-hs200-1_8v;`, the eMMC seems to be detected.

```
root@Khadas:~# dmesg | grep mmc1
[    4.685692] mmc1: new DDR MMC card at address 0001
[    4.689331] mmcblk1: mmc1:0001 CJNB4R 58.2 GiB 
[    4.693570] mmcblk1boot0: mmc1:0001 CJNB4R partition 1 4.00 MiB
[    4.699408] mmcblk1boot1: mmc1:0001 CJNB4R partition 2 4.00 MiB
[    4.704836] mmcblk1rpmb: mmc1:0001 CJNB4R partition 3 4.00 MiB, chardev (245:0)
```
But I can't operate on it, when I execute `fdisk /dev/mmcblk1` I got some errors. 
```
root@Khadas:~# dmesg | tail -n 25
[   93.658735] print_req_error: I/O error, dev mmcblk1, sector 0
[   93.662830] Buffer I/O error on dev mmcblk1, logical block 0, async page read
[   93.671399] print_req_error: I/O error, dev mmcblk1, sector 0
[   93.675583] Buffer I/O error on dev mmcblk1, logical block 0, async page read
[   93.685774] print_req_error: I/O error, dev mmcblk1, sector 0
[   93.688950] Buffer I/O error on dev mmcblk1, logical block 0, async page read
[   93.697417] print_req_error: I/O error, dev mmcblk1, sector 0
[   93.701713] Buffer I/O error on dev mmcblk1, logical block 0, async page read
[   93.708838]  mmcblk1: unable to read partition table
[   93.733383] print_req_error: I/O error, dev mmcblk1, sector 122142592
[   93.734787] print_req_error: I/O error, dev mmcblk1, sector 122142592
[   93.740597] Buffer I/O error on dev mmcblk1, logical block 15267824, async page read
[  227.949912] print_req_error: I/O error, dev mmcblk1, sector 0
[  227.955290] print_req_error: I/O error, dev mmcblk1, sector 0
[  227.959419] Buffer I/O error on dev mmcblk1, logical block 0, async page read
[  227.967973] print_req_error: I/O error, dev mmcblk1, sector 0
[  227.972189] Buffer I/O error on dev mmcblk1, logical block 0, async page read
[  227.981920] print_req_error: I/O error, dev mmcblk1, sector 0
[  227.985378] Buffer I/O error on dev mmcblk1, logical block 0, async page read
[  227.993934] print_req_error: I/O error, dev mmcblk1, sector 0
[  227.998161] Buffer I/O error on dev mmcblk1, logical block 0, async page read
[  228.005237]  mmcblk1: unable to read partition table
[  228.027150] print_req_error: I/O error, dev mmcblk1, sector 122142592
[  228.028651] print_req_error: I/O error, dev mmcblk1, sector 122142592
[  228.034350] Buffer I/O error on dev mmcblk1, logical block 15267824, async page read```

numbqq, 2019-02-14:
Update:

When I remove patch https://git.kernel.org/pub/scm/linux/kernel/git/ulfh/mmc.git/commit/?h=fixes&id=71e3e00c96efa851b2d5d3a3731711f9cc958b5b

it seems work for me.

superna9999, 2019-02-14:
Interesting thanks for testing

@jeromebrunet what do you think ?

jeromebrunet, 2019-02-14:
I think MMC is a pain :(

The patch mentioned is actually useful to some other devices, same for signal resampling. We can't just get rid of it. Unfortunately, when posted, there was an RFT on these patches but nobody cared at that time.

I guess we will have to revisit the MMC on as many devices as possible (including g12) but this will take time

superna9999, 2019-02-21:
@numbqq we are working to find an acceptable working state over different SoC generation...

dsir-minute, 2019-06-16:
@superna9999  @numbqq  : has something good enough happened regarding EMMC support in mainline kernel 5.0.4 and later versions ? thanks!

superna9999, 2019-09-14:
@numbqq did you try with recent Linux releases ? 5.2 and 5.3 should work fine now

numbqq, 2019-09-16:
Hello Neil,

Yes, seems work fine with latest mainline kernel. Thanks.

closed 2019-09-16 by numbqq