superna9999/meta-meson #60: Flashing image to eMMC

damdinsharav, 2019-02-22:
Hello,

I've successfully compiled P281 image.  Now how do I flash image to emmc?

Can we use Amlogic USB burning tool?

damdinsharav, 2019-04-03:
Can anybody guide me how to boot into compiled image. I flashed with etcher but it just boots directly to Android firmware.

urgodfather, 2019-06-04:
Usually there are a few ways to install to NAND. Depending on what you have compiled, you have to either A) boot into a custom recovery and wipe then flash. see TWRP B) once booted, you can dd the image to the NAND.

damdinsharav, 2019-07-21:
> Usually there are a few ways to install to NAND. Depending on what you have compiled, you have to either A) boot into a custom recovery and wipe then flash. see TWRP B) once booted, you can dd the image to the NAND.

I have compiled "amlogic-image-sato" image for Amlogic-p281.

I like to boot from it. It can be from SD card, not necessarily from EMMC. 
My current device is https://www.cnx-software.com/2017/08/24/x96-mini-amlogic-s905w-android-tv-box-sells-for-25-and-up/.

Thanks in advance.

closed 2019-10-13 by damdinsharav