superna9999/meta-meson #70: Amlogic s905w processor is not booting

vasujois, 2019-10-13:
Hi,

I am trying to boot Amlogic s905w processor based board. For that i followed the same steps which is mentioned in the README document.

I did the set up of yocto and built the image successfully as you mentioned in the README. I have flashed the built image on sd card. 

But, the board is not booting.  Logs also not coming on HDMI monitor. 

Can you please help me to resolve this problem. 

Thanks in advance.
VJ

damdinsharav, 2019-10-13:
Hi, what is your board? 
I am just learning but from my understanding amlogic-p281 has should boot from vendor-boot, meaning it's not included in image (wic) file.

vasujois, 2019-10-13:
amlogic s905w...

damdinsharav, 2019-10-13:
That's a SoC.

superna9999, 2019-10-13:
Hi please answer to all the question in the issue template please, otherwise I can’t answer correctly

vasujois, 2019-10-26:
I am using a customized board which consists amlogic s905w processor. 

Can any one please let me know how to build yocto image.

vasujois, 2019-12-27:
The same issue is consolidated in new topic

closed 2019-12-27 by vasujois