superna9999/meta-meson #78: python2 support on OE master

twoerner, 2020-02-12:
python2 support has been removed from upstream OE/YP master and moved to a separate layer (https://git.openembedded.org/meta-python2). The fip things required to put u-boot-meson-gx together (I'm using libretech-cc) are python2 .pyc bytecode. It would appear that someone has un-compiled the bytecode to a regular python script here: https://github.com/repk/gxlimg If that could run under python3, or if the fip things could be recompiled to be python3 bytecode, then meta-meson would not need to depend on meta-python2.

Now, in order to successfully build libretech-cc on master, one needs:
- meta-meson (master5b1df5111a27bec3a09bfcf5e2948d5320d6e133, plus the two fixes here: https://github.com/superna9999/meta-meson/pull/77)
- openembedded-core (master:44a4ac2294da0f53cbbfabc7ece836fe97f4d3f7)
- meta-python2 (master:ac385337b357ace532bbdf24fc4ff09e7b29f4b6)
- meta-openembedded/meta-oe (master:611b0c707aa889b1b3880687334b6dd0fd69479b) (this is a dependency of meta-python2)

superna9999, 2020-02-13:
Answered on #77 Libreelec now uses https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/tools/amlogic-boot-fip/sources/acs_tool.py

superna9999, 2020-04-23:
Fixed

closed 2020-04-23 by superna9999