superna9999/meta-meson #79: Failure to build u-boot for GXM

diego0815, 2020-05-24:
I tried building amlogic-image-sato for amlogic_q200 and failed for u-boot:

`|   gcc  -isystem/home/diego1/Downloads/amlogic-bsp/build/tmp/work/amlogic_q200-poky-linux/u-boot-amlogic/1_v2015.01+gitAUTOINC+ffd40ecd0e-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/diego1/Downloads/amlogic-bsp/build/tmp/work/amlogic_q200-poky-linux/u-boot-amlogic/1_v2015.01+gitAUTOINC+ffd40ecd0e-r0/recipe-sysroot-native/usr/lib                         -L/home/diego1/Downloads/amlogic-bsp/build/tmp/work/amlogic_q200-poky-linux/u-boot-amlogic/1_v2015.01+gitAUTOINC+ffd40ecd0e-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/diego1/Downloads/amlogic-bsp/build/tmp/work/amlogic_q200-poky-linux/u-boot-amlogic/1_v2015.01+gitAUTOINC+ffd40ecd0e-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/diego1/Downloads/amlogic-bsp/build/tmp/work/amlogic_q200-poky-linux/u-boot-amlogic/1_v2015.01+gitAUTOINC+ffd40ecd0e-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/diego1/Downloads/amlogic-bsp/build/tmp/work/amlogic_q200-poky-linux/u-boot-amlogic/1_v2015.01+gitAUTOINC+ffd40ecd0e-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/diego1/Downloads/amlogic-bsp/build/tmp/work/amlogic_q200-poky-linux/u-boot-amlogic/1_v2015.01+gitAUTOINC+ffd40ecd0e-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/diego1/Downloads/amlogic-bsp/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2  -o scripts/kconfig/conf scripts/kconfig/conf.o scripts/kconfig/zconf.tab.o
| mkdir -p include/config include/generated
| scripts/kconfig/conf --silentoldconfig Kconfig
| ***
| *** Configuration file ".config" not found!
| ***
| *** Please run some configurator (e.g. "make oldconfig" or
| *** "make menuconfig" or "make xconfig").
| ***
| make[3]: *** [/home/diego1/Downloads/amlogic-bsp/build/tmp/work/amlogic_q200-poky-linux/u-boot-amlogic/1_v2015.01+gitAUTOINC+ffd40ecd0e-r0/git/scripts/kconfig/Makefile:37: silentoldconfig] Error 1
| make[2]: *** [/home/diego1/Downloads/amlogic-bsp/build/tmp/work/amlogic_q200-poky-linux/u-boot-amlogic/1_v2015.01+gitAUTOINC+ffd40ecd0e-r0/git/Makefile:477: silentoldconfig] Error 2
| make[1]: *** No rule to make target 'include/config/auto.conf', needed by 'include/config/uboot.release'.  Stop.
| make: *** [Makefile:147: sub-make] Error 2
| make: Leaving directory '/home/diego1/Downloads/amlogic-bsp/build/tmp/work/amlogic_q200-poky-linux/u-boot-amlogic/1_v2015.01+gitAUTOINC+ffd40ecd0e-r0/git'
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/diego1/Downloads/amlogic-bsp/poky/../meta-meson/recipes-bsp/u-boot/u-boot-amlogic_v2015.01.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4771 tasks of which 4762 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/diego1/Downloads/amlogic-bsp/poky/../meta-meson/recipes-bsp/u-boot/u-boot-amlogic_v2015.01.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
`
Build environment is Debian bullseye/sid I tried first all exactly as described in the readme, then tried master, with similar results.

Is there any obvious mistake I did?

superna9999, 2020-05-26:
Which branch of meta-meson did you build ?

diego0815, 2020-05-26:
Hi Neil

First zeus:
`amlogic-bsp$ git clone https://github.com/superna9999/meta-meson.git meta-meson -b zeus`

then master, both failed with similar error messages.

All in all I followed this readme https://github.com/superna9999/meta-meson/blob/zeus/README.md

superna9999, 2020-05-26:
I can reproduce, thanks, let me find out what's the issue is

diego0815, 2020-05-26:
Super thanks, unfortunately I wasn't able to track it further down, as it's my first excursion into yocto.

closed 2021-09-23 by superna9999