superna9999/meta-meson #81: linux-yocto-meson64 doesn't use linux-yocto

m5p3nc3r, 2020-05-29:
**Describe the bug**
Because of the naming convention 'linux-yocto-meson64_<version>.bb, I would have expected the 64bit kernel build for meson to conform to the linux-yocto way of doing things.
It turns out the the 32bit version does, but the 64bit version breaks the convention.

**Build Context**
 - meta-meson zeus

**Expected behaviour**
I would have hoped to be able to use the yocto-linux methods for modifying kernel configuration etc

**Additional context**
I am assuming there is a reason for not inheriting from linux-yocto, but this does not appear to be documented anywhere.
Is there any way we can have a linux-meson64 for custom build and yocto-linux-meson64 for a Yocto compliant build process?

superna9999, 2020-05-29:
The main reason was :
- beeing able to have defconfigs instead of full kernel configs
- removing autoinc feature breaking out-of-tree patches

You can find the changes from @koenkooi like https://github.com/superna9999/meta-meson/commit/4302ff8dc7ca264ea7bb72fcb93b47ab20bab0c5#diff-9b732ab86ef3483bb0e54684c28fd84b

But feel free to rename yocto-linux-meson64 into linux-meson64 and add a vanilla yocto-linux-meson64 not having any out-of-tree patches and using upstream defconfig in a pull request.

koenkooi, 2020-05-29:
And it saves a few minutes of grinding through the ‘yocto’ way of patching. On my system it saved about 5 minutes between unpacking and configure.

m5p3nc3r, 2020-05-29:
Hey guys

Thanks for the prompt response.  I'll have a go at creating an upstream/linux-yocto compatible configuration and post the patches here.  But I expect my yocto-fu might not be as good as yours!

m5p3nc3r, 2020-06-03:
I have a patch set available now that does the following
- Renames recipes for linux-yocto-meson64 => linux-meson64
- Updates preferred provider from linux-yocto-meson64 => linux-meson64 (keeping default behaviour)
- Creates a linux-yocto_%.bbappend and kmeta configuration for basic vim3 booting

Code can be found here https://github.com/m5p3nc3r/meta-meson/tree/linux-yocto_rework

Currently tested and working include
- Boot from sd-card
- console
- USB (validated with mass storage device)
- ethernet (validated with ping)

If you are happy with this as a first pass to support linux-yocto, I will create a pull request.

superna9999, 2020-06-04:
I added a comment at https://github.com/m5p3nc3r/meta-meson/commit/ec669641447a0d9b5331187467b7efc7e5bc43df#commitcomment-39634687

You should add a generic meson64 kmeta to cover all boards.

m5p3nc3r, 2020-06-04:
Hey @superna9999, thanks for the feedback.
I have renamed kmeta-khadas as per your recommendation.
I'm not sure what to do about the KERNEL_CLASSES issue, will have a think about it but my gut feel is that it is a feature of the way bitbake is working under the hood?  Do you have any ideas?

superna9999, 2020-06-04:
you could eventually set the KERNEL_CLASSES in the conf/machine/include main files and disable it for board not using uImage ?

m5p3nc3r, 2020-06-05:
So adding:
KERNEL_CLASSES_khadas-vim3 ?= "kernel-uimage-meson"
to
conf/machine/include/amlogic-a311d.inc

should be ok?  Or should it be a generic:
KERNEL_CLASSES ?= "kernel-uimage-meson"

m5p3nc3r, 2020-06-06:
pull request created #85

closed 2020-06-16 by superna9999