superna9999/meta-meson #82: wifi-fw-2016-08-18-5f5fae7239 fetch failure

dagmcr, 2020-05-29:
**Describe the bug**
`wifi-fw-2016-08-18-5f5fae7239` build fails when fetching the tarball file:

```
ERROR: wifi-fw-2016-08-18-5f5fae7239-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; unset _PYTHON_SYSCONFIGDATA_NAME; export SSH_AUTH_SOCK="/ssh-agent"; export PATH="/workdir/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/workdir/poky/scripts:/workdir/build/tmp/work/aarch64-poky-linux/wifi-fw/2016-08-18-5f5fae7239-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/workdir/build/tmp/work/aarch64-poky-linux/wifi-fw/2016-08-18-5f5fae7239-r0/recipe-sysroot/usr/bin/crossscripts:/workdir/build/tmp/work/aarch64-poky-linux/wifi-fw/2016-08-18-5f5fae7239-r0/recipe-sysroot-native/usr/sbin:/workdir/build/tmp/work/aarch64-poky-linux/wifi-fw/2016-08-18-5f5fae7239-r0/recipe-sysroot-native/usr/bin:/workdir/build/tmp/work/aarch64-poky-linux/wifi-fw/2016-08-18-5f5fae7239-r0/recipe-sysroot-native/sbin:/workdir/build/tmp/work/aarch64-poky-linux/wifi-fw/2016-08-18-5f5fae7239-r0/recipe-sysroot-native/bin:/workdir/poky/bitbake/bin:/workdir/build/tmp/hosttools"; export HOME="/home/pokyuser"; /usr/bin/env wget -t 2 -T 30 --passive-ftp --no-check-certificate -P /workdir/build/downloads 'http://openlinux.amlogic.com:8000/download/ARM/wifi/wifi-fw-2016-08-18-5f5fae7239.tar.gz' --progress=dot -v failed with exit code 8, output:
--2020-05-29 16:31:43--  http://openlinux.amlogic.com:8000/download/ARM/wifi/wifi-fw-2016-08-18-5f5fae7239.tar.gz
Resolving openlinux.amlogic.com (openlinux.amlogic.com)... 120.25.202.108
Connecting to openlinux.amlogic.com (openlinux.amlogic.com)|120.25.202.108|:8000... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-05-29 16:31:46 ERROR 404: Not Found.
```

So, repo `http://openlinux.amlogic.com:8000/download/ARM` no longer exits.

I was trying to find `wifi-fw-2016-08-18-5f5fae7239.tar.gz` file but I couldn't find it so, I was looking for the files the recipe was trying to install and I end up looking in this 3 repos:

*  https://github.com/armbian/firmware
*  https://github.com/hardkernel/buildroot_platform_hardware_amlogic_wifi.git
*  git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git

And here the files matching the names on the recipe:

*  file: `fw_bcm4356a2_ag.bin`

    *   `efc2555f3792101082a51714be4b4ba4  ./armbian-firmware/rkwifi/fw_bcm4356a2_ag.bin`
    *  `1e7dcba997e5dc1d9970f9586eb85dfe  ./buildroot_platform_hardware_amlogic_wifi/bcm_ampak/config/4356/fw_bcm4356a2_ag.bin`

*  file: `fw_bcm4356a2_ag.bin` renamed to: `brcmfmac4356-sdio.bin`:
    *  `ca63a118d10e69d0f84d8e06bbbe9d4f  ./armbian-firmware/brcm/brcmfmac4356-sdio.bin`
    *  `57853b6ac141f6d8d5774206436d9c55  ./linux-firmware/brcm/brcmfmac4356-sdio.bin`

*  file: `nvram_ap6356.txt`:
    *  `bd37ec8da4d46816aba85d018dbbbec8  ./armbian-firmware/rkwifi/nvram_ap6356.txt`
    *  `b7a7e2d284c5284d8f7e266cbdff5480  ./buildroot_platform_hardware_amlogic_wifi/bcm_ampak/config/4356/nvram_ap6356.txt`

*  file: `nvram_ap6356.txt` renamed to: `brcmfmac4356-sdio.txt`:
    *  `046a0d584bab0d2f774ba9f722a175f9  ./armbian-firmware/brcm/brcmfmac4356-sdio.txt`

*  file: `nvram_ap6354.txt`: 
    *    `53c14427b50297acd443bb59a9b1769b  ./armbian-firmware/rkwifi/nvram_ap6354.txt`
    *   `714f7a3796eca86f41cdf8b5216a475a  ./buildroot_platform_hardware_amlogic_wifi/bcm_ampak/config/4354/nvram_ap6354.txt`

*  file: `nvram_ap6354.txt` renamed to: `brcmfmac4354-sdio.txt`
    *  not found
    

**Build Context**
 - meta-meson zeus, master
Recipe is the same in both branches.

**Expected behavior**
Fetch with no failure.

**Additional context**
Do you have the tarball so we can upload it to the recipe  `./recipes-firmware/wifi-fw/wifi-fw/<files>` or somewhere else?
If not, is any of the files listed above the right ones?

Thanks!

superna9999, 2020-06-01:
Thanks, but this recipe is very old and the matching tarball doesn't exist anymore. I expect all these binaries are now almost into linux-firmware no, so only missing ones should be individually added to the linux-firmware recipe. I'll remove this recipe shortly.

superna9999, 2020-06-01:
You can also find some firmwares in https://github.com/LibreELEC/wlan-firmware or https://github.com/LibreELEC/brcmfmac_sdio-firmware

closed 2020-06-24 by superna9999