superna9999/meta-meson #107: Modify UBOOT_EXTLINUX_ROOT

m5p3nc3r, 2021-05-20:
On my Vim3, when booting I need to modify the UBOOT_EXTLINUX_ROOT to /dev/mmcblk1p1.  With the current recipes, it is not possible to do this in my local.conf because it has a hard assignment in conf/machine/<machine>.conf.

Would you accept a PR that changes the line

UBOOT_EXTLINUX_ROOT = "root=/dev/mmcblk0p1"
 to
UBOOT_EXTLINUX_ROOT ?= "root=/dev/mmcblk0p1"

as this would not change the default behaviour, but allow me to change the default to what I need?

If so, should I create a PR just for the vim3, or should I make the modification for all platforms that use UBOOT_EXTLINUX_ROOT?

superna9999, 2021-05-20:
sure no problem do the change for all platforms

m5p3nc3r, 2021-05-20:
Do you want it against the master branch?

superna9999, 2021-05-20:
yes, I'll push manually on the other branches

m5p3nc3r, 2021-05-20:
Created PR #108

closed 2021-09-23 by superna9999