superna9999/meta-meson #120: - nothing provides linux-firmware-bcm43430a0 needed by packagegroup-core-boot-1.0-r17.seirobotics_sei610

EzEmbedded, 2021-11-30:
**My  git branch:**

```
git clone --depth 1 git://git.yoctoproject.org/poky -b hardknott  poky
git clone https://github.com/superna9999/meta-meson.git -b hardknott 
git clone git://git.openembedded.org/meta-openembedded -b hardknott 
git clone git://git.yoctoproject.org/meta-virtualization -b hardknott 
git clone git://git.yoctoproject.org/meta-security -b hardknott 
git clone git://github.com/kraj/meta-clang -b hardknott 
```

**My local.conf:**

```
MACHINE = "seirobotics-sei610" #or "khadas-vim3l"
DL_DIR ?= "/home/yocto-crops/downloads"
BB_GENERATE_MIRROR_TARBALLS = "1"
INHERIT += "buildhistory"
BUILDHISTORY_COMMIT = "1"
BB_NUMBER_THREADS = "8"
PARALLEL_MAKE = "-j 8"
LICENSE_FLAGS_WHITELIST += " commercial_mpg123 commercial commercial_libav commercial_x264 commercial"
DISTRO_FEATURES_append = " virtualization"
PACKAGE_CLASSES ?= "package_ipk"
IMAGE_INSTALL_append += " bash docker-ce docker-ce-contrib python3 python3-docker-compose"
```
**My BBLAYERS :**
```
BBLAYERS ?= " \
  /home/yocto-crops/poky/meta \
  /home/yocto-crops/poky/meta-poky \
  /home/yocto-crops/poky/meta-yocto-bsp \
  /home/yocto-crops/meta-openembedded/meta-oe \
  /home/yocto-crops/meta-openembedded/meta-python \
  /home/yocto-crops/meta-openembedded/meta-perl \
  /home/yocto-crops/meta-openembedded/meta-networking \
  /home/yocto-crops/meta-openembedded/meta-filesystems \
  /home/yocto-crops/meta-openembedded/meta-webserver \
  /home/yocto-crops/meta-virtualization \
  /home/yocto-crops/meta-security \
  /home/yocto-crops/meta-meson \
  "

```
**The ERRO:**
```
ERROR: amlogic-image-headless-initrd-1.0-r0 do_rootfs: Could not invoke dnf. Command '/home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/amlogic-image-headless-initrd/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=info -y -c /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/amlogic-image-headless-initrd/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/amlogic-image-headless-initrd/1.0-r0/rootfs/etc/yum.repos.d --installroot=/home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/amlogic-image-headless-initrd/1.0-r0/rootfs --setopt=logdir=/home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/amlogic-image-headless-initrd/1.0-r0/temp --repofrompath=oe-repo,/home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/amlogic-image-headless-initrd/1.0-r0/oe-rootfs-repo --nogpgcheck install bash docker-ce docker-ce-contrib kernel-devicetree kernel-modules linux-firmware packagegroup-core-boot packagegroup-core-ssh-openssh python3 python3-docker-compose run-postinsts locale-base-en-us locale-base-en-gb' returned 1:
DNF version: 4.6.0
cachedir: /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/amlogic-image-headless-initrd/1.0-r0/rootfs/var/cache/dnf
Added oe-repo repo from /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/amlogic-image-headless-initrd/1.0-r0/oe-rootfs-repo
User-Agent: falling back to 'libdnf': could not detect OS or basearch
repo: using cache for: oe-repo
oe-repo: using metadata from Tue 30 Nov 2021 08:29:14 AM UTC.
Last metadata expiration check: 0:00:01 ago on Tue 30 Nov 2021 08:29:15 AM UTC.
--> Starting dependency resolution
--> Finished dependency resolution
Error: 
 Problem: conflicting requests
  - nothing provides linux-firmware-bcm43430a0 needed by packagegroup-core-boot-1.0-r17.seirobotics_sei610
(try to add '--skip-broken' to skip uninstallable packages)

```
**when MACHINE = "hardkernel-odroidc4"  pass all.**

superna9999, 2021-11-30:
Thanks, looking right now

superna9999, 2021-11-30:
Reproduced, indeed `linux-firmware-bcm43430a0` is not available on the linux-firmware version of hardknott. Fixed on `hardknott` branch

closed 2021-11-30 by superna9999

EzEmbedded, 2021-12-03:
It,s OK now! thanks superna9999!

commit afe6646f (2021-11-30): machine: amlogic-meson.inc: remove invalid linux-firmware-bcm43430a0 MACHINE_ESSENTIAL_EXTRA_RDEPENDS

Fixes #120