superna9999/meta-meson #122: ERROR: mesa-2_21.0.3-r0 do_configure: meson failed

EzEmbedded, 2021-12-03:
Hi superna9999,
I‘m adding meta-kodi(https://github.com/koenkooi/meta-kodi) to meta-meson..the branch is hardknott.
but I encountered the fellow errors:
```
ERROR: Logfile of failure stored in: /home/yocto-crops/build/tmp/work/aarch64-poky-linux/mesa/2_21.0.3-r0/temp/log.do_configure.9831
ERROR: Task (/home/yocto-crops/poky/meta/recipes-graphics/mesa/mesa_21.0.3.bb:do_configure) failed with exit code '1'
```
**In log.do_configure.9831**
```
DEBUG: Executing python function extend_recipe_sysroot
NOTE: ......
NOTE:......
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing shell function do_configure
NOTE: Executing meson -Dshared-glapi=enabled -Dgallium-opencl=disabled -Dglx-read-only-text=true -Dplatforms=wayland,x11 -Ddri-drivers= -Ddri3=enabled -Degl=enabled -Delf-tls=true -Dgallium-drivers=swrast,kmsro,virgl,lima,panfrost -Dllvm=false -Dgbm=enabled -Dgles1=enabled -Dgles2=enabled -Dopengl=true -Dosmesa=false -Dlibunwind=disabled -Dgallium-va=enabled -Dgallium-vdpau=enabled -Dvulkan-drivers= -Dgallium-xa=disabled -Dgallium-xvmc=disabled...
The Meson build system
Version: 0.57.1
Source dir: /home/yocto-crops/build/tmp/work/aarch64-poky-linux/mesa/2_21.0.3-r0/mesa-21.0.3
Build dir: /home/yocto-crops/build/tmp/work/aarch64-poky-linux/mesa/2_21.0.3-r0/build
Build type: cross build
Program python3 found: YES (/home/yocto-crops/build/tmp/work/aarch64-poky-linux/mesa/2_21.0.3-r0/recipe-sysroot-native/usr/bin/python3-native/python3)
Project name: mesa
Project version: 21.0.3
C compiler for the host machine: aarch64-poky-linux-gcc -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/yocto-crops/build/tmp/work/aarch64-poky-linux/mesa/2_21.0.3-r0/recipe-sysroot (gcc 10.2.0 "aarch64-poky-linux-gcc (GCC) 10.2.0")
C linker for the host machine: aarch64-poky-linux-gcc -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/yocto-crops/build/tmp/work/aarch64-poky-linux/mesa/2_21.0.3-r0/recipe-sysroot ld.bfd 2.36.1.20210209
C++ compiler for the host machine: aarch64-poky-linux-g++ -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/yocto-crops/build/tmp/work/aarch64-poky-linux/mesa/2_21.0.3-r0/recipe-sysroot (gcc 10.2.0 "aarch64-poky-linux-g++ (GCC) 10.2.0")
C++ linker for the host machine: aarch64-poky-linux-g++ -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/yocto-crops/build/tmp/work/aarch64-poky-linux/mesa/2_21.0.3-r0/recipe-sysroot ld.bfd 2.36.1.20210209
C compiler for the build machine: gcc (gcc 7.5.0 "gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0")
C linker for the build machine: gcc ld.bfd 2.30
C++ compiler for the build machine: g++ (gcc 7.5.0 "g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0")
C++ linker for the build machine: g++ ld.bfd 2.30
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: aarch64
Host machine cpu: aarch64
Target machine cpu family: aarch64
Target machine cpu: aarch64
Found CMake: NO
Run-time dependency clang found: NO (tried cmake)

../mesa-21.0.3/meson.build:546:4: ERROR: Problem encountered: VDPAU state tracker requires at least one of the following gallium drivers: r300, r600, radeonsi, nouveau.

```

superna9999, 2021-12-03:
Hi, for which MACHINE ?

superna9999, 2021-12-03:
The problem is at https://github.com/koenkooi/meta-kodi/blob/ac5e11304a0bfd3fa2ba87d6dccdd4ce168256f6/classes/vpu.bbclass#L2

What image do you build ? Try building an image without x11 configured

EzEmbedded, 2021-12-06:
my config:
```
MACHINE = "seirobotics-sei610"
LICENSE_FLAGS_WHITELIST += " commercial commercial_mpg123 commercial_libav commercial_x264"
DISTRO_FEATURES_append = " virtualization systemd pam"
DISTRO_FEATURES_remove = " x11"
IMAGE_INSTALL_append += " kodi"

```
then run:
`bitbake amlogic-image-headless-initrd`

but the error is same.

superna9999, 2021-12-07:
Reproduced, thx

superna9999, 2021-12-07:
With:
```
DISTRO_FEATURES_remove = "x11"
```
and
```
MACHINE=khadas-vim3l bitbake mesaMACHINE=khadas-vim3l bitbake mesa
```

I get:
```
| ../mesa-21.0.3/meson.build:720:4: ERROR: Problem encountered: VA state tracker requires at least one of the following gallium drivers: r600, radeonsi, nouveau.
```

Which is the same but different, now VA is enabled for `opengl` distro feature :-/

So really this should be fixed in `meta-kodi` instead

superna9999, 2021-12-07:
VA & VDPAU as exclusively x86 features

EzEmbedded, 2021-12-08:
I have tryed meta-kodi whith meta-raspberrypi,The error is same as meta-meson:mesa-21.0.3/meson.build:546:4: ERROR: Problem encountered: VDPAU state tracker requires at least one of the following gallium drivers: r300, r600, radeonsi, nouveau.

closed 2021-12-08 by EzEmbedded