superna9999/meta-meson #123: WARNING: /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.38771:317 exit 1 from 'cp ${FIPDIR}/bl31.img ${DESTDIR}/bl31.img'

EzEmbedded, 2021-12-08:
**conf:**
```
MACHINE="seirobotics-sei610"
image:amlogic-image-sato
branch:hardknott
```
**error:**
```
WARNING: /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.38771:317 exit 1 from 'cp ${FIPDIR}/bl31.img ${DESTDIR}/bl31.img'
WARNING: Backtrace (BB generated script): 
	#1: deploy_g12a, /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.38771, line 317
	#2: do_deploy, /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.38771, line 301
	#3: main, /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.38771, line 532
ERROR: ExecutionError('/home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.38771', 1, None, None)

```

superna9999, 2021-12-08:
I can't reproduce

EzEmbedded, 2021-12-09:
**confg:**
```
branch   :hardknott
MACHINE="seirobotics-sei610"
PACKAGECONFIG_append_pn-chromium = " component-build proprietary-codecs use-vaapi"
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-good = "vpx wavpack libv4l2"
LICENSE_FLAGS_WHITELIST += " commercial commercial_mpg123 commercial_libav commercial_x264"
DISTRO_FEATURES_append = " virtualization systemd pam opengl egl"
```
```
BBLAYERS ?= " \
  /home/yocto-crops/poky/meta \
  /home/yocto-crops/poky/meta-poky \
  /home/yocto-crops/poky/meta-yocto-bsp \
  /home/yocto-crops/meta-openembedded/meta-oe \
  /home/yocto-crops/meta-openembedded/meta-python \
  /home/yocto-crops/meta-openembedded/meta-perl \
  /home/yocto-crops/meta-openembedded/meta-multimedia \
  /home/yocto-crops/meta-openembedded/meta-networking \
  /home/yocto-crops/meta-openembedded/meta-filesystems \
  /home/yocto-crops/meta-openembedded/meta-webserver \
  /home/yocto-crops/meta-clang \
  /home/yocto-crops/meta-python2 \
  /home/yocto-crops/meta-virtualization \
  /home/yocto-crops/meta-security \
  /home/yocto-crops/meta-meson \
  /home/yocto-crops/meta-browser/met
```
**when run bitbake amlogic-image-headless-initrd/amlogic-image-headless-sd , all succeeded**
**but** 
**when run bitbake amlogic-image-sato ,the errors are:**
```
DEBUG: Executing python function sstate_task_prefunc
DEBUG: Python function sstate_task_prefunc finished
DEBUG: Executing shell function do_deploy
cp: cannot stat '/home/yocto-crops/build/tmp/deploy/images/seirobotics-sei610/fip//bl31.img': No such file or directory
WARNING: /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.37708:353 exit 1 from 'cp ${FIPDIR}/bl31.img ${DESTDIR}/bl31.img'
WARNING: Backtrace (BB generated script): 
	#1: deploy_g12a, /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.37708, line 353
	#2: do_deploy, /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.37708, line 301
	#3: main, /home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.37708, line 532
ERROR: ExecutionError('/home/yocto-crops/build/tmp/work/seirobotics_sei610-poky-linux/u-boot-meson-gx/1_2021.10-r0/temp/run.do_deploy.37708', 1, None, None)
```
**when run bitbake amlogic-image-headless-initrd , all succeeded**

superna9999, 2021-12-14:
I haven't managed to reproduce it so far... can you test on `hardknott-next ` ?

EzEmbedded, 2021-12-15:
Hi superna9999,you are right,I test severl times and can't reproduce it too,in branch hardknott,thks!

closed 2021-12-15 by EzEmbedded