superna9999/meta-meson #127: Integrate with meta-swupdate

superna9999, 2021-12-14:
TODO in https://github.com/rauc/meta-rauc-community

Contributions are welcome !

graugans, 2022-05-22:
I guess the link should point to something like: [meta-swupdate-boards](https://github.com/sbabic/meta-swupdate-boards). What is your preferred mechanism for swupdate? A/B image or recovery?

superna9999, 2022-05-22:
> I guess the link should point to something like: [meta-swupdate-boards](https://github.com/sbabic/meta-swupdate-boards). What is your preferred mechanism for swupdate? A/B image or recovery?

Thanks indeed, I’ll update the link.

I have no strong opinion on the mechanism, does it change the implementation a lot ? Can it be selected with a variable ?

graugans, 2022-05-22:
I have to dig into `wic` it has been a while. The A/B image comes with the benefit that you can easily update from the current system. But it comes with the downside of half of the available flash memory. It may also become tricky with unknown eMMC sizes. Which is a typical situation for Amlogic consumer hardware I guess. The recovery approach needs a reboot into recovery for the update.

I'll prepare something.