superna9999/meta-meson #131: Only compiles on x86 host

m5p3nc3r, 2022-05-20:
**Describe the bug**
This is a know issue, as it is clearly marked in recipes-bsp/trusted-firmware-a/amlogic-boot-fip.bb
But as infrastructure is being enabled on aarch64 hosts, it would be good to get ahead of the problem and track a solution.
The binary tools supplied as a part of https://github.com/LibreELEC/amlogic-boot-fip are x86 only, so its not possible to build images on an arm host.

**Build Context**
Host machine is aarch64

**Expected behavior**
The image should build normally

**System (please complete the following information):**
 - Host OS: Fedora
 - Version 36
 - Architecture aarch64
 - Specific packages: amlogic-boot-fip

**Additional context**
This issue is clearly documented [here](https://github.com/superna9999/meta-meson/blob/f3c65574a5c9f02e6c9a9223d49f92b298561b66/recipes-bsp/trusted-firmware-a/amlogic-boot-fip.bb#L8)

It looks like there may be options to build without using the binary only amlogic tools.  Extract of details here, source from    https://github.com/LibreELEC/amlogic-boot-fip

- https://github.com/afaerber/meson-tools (GXBB, GXL & GXM only)
- https://github.com/repk/gxlimg (GXBB, GXL, GXM & AXG only)
- https://github.com/angerman/meson64-tools (developed for G12B, should work on G12A & SM1)

superna9999, 2022-05-22:
Thanks, I was thinking about it. Did you try adding support for it ?

it may be help but I opened a PR to use gxlimg tools: https://github.com/LibreELEC/amlogic-boot-fip/pull/4

superna9999, 2022-05-30:
If it can help, I pushed a gxlimg recipe in 64a8446

superna9999, 2022-05-30:
And meson-tools recipe in 70118f9

GXBB isn't handled by gxlimg, but only by meson-tools.

AMuszkat, 2024-12-08:
Any solutions for building in aarch64 host ? have anyone tested it ?

AMuszkat, 2024-12-11:
Tested this to build bananapi-bpi-cmio machine. It works with a small correction in amlogic-boot-fip.bb. [https://github.com/AMuszkat/meta-meson/commit/5da45aa051cb6c8d18cb4f0db87e49fbd2131ebf](url)

Can't find why a "space+gxlimg" is needed instead of just "gxlimg"

Don't see the need for the patch anymore.

Still haven't tried to boot the hardware.