superna9999/meta-meson #135: Dokumentation improvement

graugans, 2022-06-02:
**Is your feature request related to a problem? Please describe.**
My U-Boot and kernel hacking skills became a bit rusty over the last few years, I guess there are other people like me who would appreciate some documentation on how to use the for example the initrd image.

**Describe the solution you'd like**
I would be more than happy to provide the information I have collected so far.

**Describe alternatives you've considered**
I am not sure if you prefer this to be in the wiki or shall this be part of a `*.md` file in a doc folder? Or shall it be directed to the linux-meson website?

**Additional context**

To use the `amlogic-image-headless-initrd` image, a little bit of preparation is needed. To boot the initial ramdisk image we do have to load the Linux Kernel, a device tree blob and the initrd image into RAM first. There are multiple ways to do so. In this example we do use the `tftfp` U-Boot command to load the image over the network from a development host. 

```
tftp ${loadaddr} uImage
tftp ${dtb_mem_addr} meson-sm1-x96-max-plus.dtb
tftp ${initrd_high} amlogic-image-headless-initrd-amediatech-x96-max-plus.cpio.xz
setenv bootargs "console=ttyAML0,115200 panic=3 root=/dev/ram0 rw  initrd=0xdf800000,8M ramdisk_size=8192"
```

graugans, 2022-06-02:
Is there any particular reason why you prefer `*.cpio.gz` over `*.cpio.gz.u-boot`? The usage becomes a bit easier IMHO.

```
tftp ${loadaddr} uImage
tftp ${dtb_mem_addr} meson-sm1-x96-max-plus.dtb
tftp ${initrd_high} amlogic-image-headless-initrd-amediatech-x96-max-plus.cpio.xz.u-boot
bootm ${loadaddr} ${initrd_high} ${dtb_mem_addr}
```

superna9999, 2022-06-02:
@graugans thankx, indeed it would be a great addition to documentation, in can totally be a bunch of md file in a new doc folder or RST files we could build into a Sphinx generated static website.
Please submit something so we can start discussing what's the best form to have.

For the choice of cpio.gz, I use .cpio to feed Lava that then generates a .u-boot, but there's no reason not to have a .u-boot also generated, so please submit a change for that.

graugans, 2022-06-02:
Another thing which may need some thoughts is the fact that the linux-firmware package blows up the initrd image size from 19MB to 235MB.

superna9999, 2022-06-02:
@graugans please create a new issue for that one

closed 2022-06-07 by superna9999