superna9999/meta-meson #139: s905-autoscript question

graugans, 2022-06-07:
From my understanding the packages `[s905-autoscript](https://github.com/superna9999/meta-meson/tree/kirkstone/recipes-bsp/s905-autoscript)` and [s905-autoscript-multiboot](https://github.com/superna9999/meta-meson/tree/kirkstone/recipes-bsp/s905-autoscript-multiboot) are intended to support the vendor U-Boot boot process. Like described in the [README](https://github.com/superna9999/meta-meson#selecting-machine-to-build-for)

> This layer aims to support all mainline supported boards, but some of them have a vendor U-Boot loaded on the eMMC and can only be booted by pressing the "update" button.

But when I do flash the image [amlogic-image-headless-sd](https://github.com/superna9999/meta-meson/blob/kirkstone/recipes-core/images/amlogic-image-headless-sd.bb) the file `aml_autoscript` is missing on the boot partition. This is added by [amlogic-s9xxx.inc] (https://github.com/superna9999/meta-meson/blob/kirkstone/conf/machine/include/amlogic-s9xxx.inc) which is only included in [amlogic-s9xxx.conf](https://github.com/superna9999/meta-meson/blob/master/conf/machine/amlogic-s9xxx.conf) which looks like a base for board which do run a mainline U-Boot and not a vendor one.

Or did I understood it wrong?

superna9999, 2022-06-07:
Those are installed in https://github.com/superna9999/meta-meson/blob/f3c65574a5c9f02e6c9a9223d49f92b298561b66/conf/machine/include/amlogic-legacy-boot.inc#L7 

This boot scheme is selected in `<board>-sdboot.conf` machine names, I didn't finish adding the `-sdboot.conf` variants for all board.

graugans, 2022-06-07:
Yes, you are right. I was only search half way. I am not sure if this is may related to my set-up but when checking the `amlogic-image-headless-sd.env` file in the image folder the variable `IMAGE_BOOT_FILES` only contains the device tree entry.

I will investigate this further.

graugans, 2022-06-07:
Looks like the order is important. 
```
require conf/machine/include/amlogic-legacy-boot.inc
require conf/machine/include/amlogic-s905x2.inc
require conf/machine/include/amediatech-x96-max-dtb.inc
```
works for me.

superna9999, 2022-06-13:
Interesting, it doesn't make sense, could you provide a fix in a PR ?

graugans, 2022-06-20:
Sorry I was a few days offline, I will do more investigation. But this week will be loaded with other stuff.