superna9999/meta-meson #142: khadas-vim3: do_kernel_configcheck: config analysis failed

nguyenlkdn, 2022-09-05:
**Describe the bug**
```
Build Configuration:
BB_VERSION           = "1.48.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-22.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "khadas-vim3"
DISTRO               = "poky"
DISTRO_VERSION       = "3.2.4"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "gatesgarth:13143ea85a1ab7703825c0673128c05845b96cb5"
meta-meson           = "gatesgarth:a6879af3f9e521a9505232b2b0af6834e7f0e2c7"

NOTE: Executing Tasks
ERROR: linux-yocto-dev-5.10-rc++gitAUTOINC+773b575c4e_7544e2ee70-r0 do_kernel_configcheck: config analysis failed: 
ERROR: Logfile of failure stored in: /mnt/data/yocto/amlogic-bsp/build/tmp/work/khadas_vim3-poky-linux/linux-yocto-dev/5.10-rc++gitAUTOINC+773b575c4e_7544e2ee70-r0/temp/log.do_kernel_configcheck.3546255
ERROR: Task (/mnt/data/yocto/amlogic-bsp/poky/meta/recipes-kernel/linux/linux-yocto-dev.bb:do_kernel_configcheck) failed with exit code '1'
NOTE: Tasks Summary: Attempted 558 tasks of which 557 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /mnt/data/yocto/amlogic-bsp/poky/meta/recipes-kernel/linux/linux-yocto-dev.bb:do_kernel_configcheck
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

**Build Context**
 - meta-meson "gatesgarth:a6879af3f9e521a9505232b2b0af6834e7f0e2c7"
 - Yocto Project Poky / oe-core branch "gatesgarth:13143ea85a1ab7703825c0673128c05845b96cb5"
 - bitbake command bitbake amlogic-image-sato

**Expected behavior**
amlogic-image-sato should be created

**System (please complete the following information):**
 - Host OS: Ubuntu
 - Version: 22.04
 - Architecture: x86_64
 - Specific packages: gcc (Ubuntu 10.3.0-15ubuntu1) 10.3.0

**Additional context**
Add any other context about the problem here.

superna9999, 2022-09-05:
Hi, Gatesgarth is EOL is since May 2021, did you try Kirkstone instead ?

nguyenlkdn, 2022-09-05:
Thank for your comment, I tried on Kirkstone branch successfully.

closed 2022-09-05 by nguyenlkdn