superna9999/meta-meson #159: Can't enable SPI dev on khadas-vim3

jjichen, 2024-09-04:
I'm using machine "khadas-vim3" with the image type "core-image-base" in this project.

However, I'm having trouble getting SPI to work. I've enabled SPI support in the kernel by setting SPI_SPIDEV to y. I tried to add the SPI device node &spicc1 to the device tree file meson-khadas-vim3.dtsi (according to the [device tree of khadas ubuntu image](https://forum.khadas.com/t/edit-device-tree-and-active-spi-with-fenix-vim3/8325)), but I'm not seeing the expected SPI device files (/dev/spidev2.1 or /dev/spidev1.1) in the /dev directory.

Do you have any suggestions on what might be going wrong or additional steps I should take to troubleshoot this issue? Thanks a lot.

superna9999, 2024-09-04:
@jjichen yes, because `compatible = "linux,spidev";` is no more supported into mainline linux, but you can use any compatible you find in the file `drivers/spi/spidev.c` like `semtech,sx1301`

superna9999, 2024-09-04:
see https://elixir.bootlin.com/linux/v6.11-rc6/source/drivers/spi/spidev.c#L718

jjichen, 2024-09-04:
Thanks a lot, it works with `compatible = "semtech,sx1301";` I have `/dev/spidev0.0` in /dev directory